bulletphysics/bullet3 #1521: Problem running App_ExampleBrowser on OSX

aravindsiv, 2018-01-19:
Hi, 

I'm trying to get the KUKA VR demo set up. Since I have to run a SHARED_MEMORY physics server before loading the robot and connecting to the VR controller, I decided to build the latest GitHub version of bullet as @erwincoumans described in #903.

After I finish running ```build_cmake_pybullet_double.sh``` and tried to run the ```App_ExampleBrowser``` executable, I got the following error on my MacBook:

```
Vendor: Intel Inc.
Renderer: Intel(R) Iris(TM) Graphics 6100
Version: 4.1 INTEL-10.25.17
GLSL: 4.10
b3Printf: Selected demo: Basic Example
Bus error: 10
```

Any idea what the problem could be? Alternatively, would it be possible for me to run a SHARED_MEMORY physics server through some other script for my particular use case?

Any kind of help would be appreciated. Thanks in advance.

JanMatas, 2018-01-24:
I can confirm I experience the same issue. It must have been introduced in some recent commits, because if I check out an older version (I tested 7118470eb6e4da60ae8c3009655bb2c3548b1c98) it works perfectly.

erwincoumans, 2018-01-24:
Yes, it was a regression, thanks for reporting.
It will be fixed with this upcoming PR:
https://github.com/bulletphysics/bullet3/pull/1524

erwincoumans, 2018-01-24:
Note that VR has only been tested on Windows, but it may work on Mac OSX. Please report back if/when you tried it.

aravindsiv, 2018-01-24:
Thanks for the fix, @erwincoumans . I'll report back once I've tried it.

closed 2018-01-25 by erwincoumans