bulletphysics/bullet3 #1529: How to get the dynamic equation?

foolyc, 2018-01-26:
[image]

how to get the matrix  M/N/S/Jc etc. after loading a  robot xml file?

erwincoumans, 2018-01-26:
PyBullet provides a function to calculate the mass matrix M, calculateMassMatrix, and Jacobian, calculateJacobian. There is an example for calculateJacobian here:
https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/jacobian.py

There is also a calculateInverseDynamics, in case you want it, see 
https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/inverse_dynamics.py

Please provide more detail what else you need exactly.

foolyc, 2018-01-29:
well, 'N' stands for centripetal, Coriolis and gravity forces, '\tao' stands for user force, '\lamda' stands for constraint forces
i want to use this matrix to do inverse dynamic QP optimization for control purpose.

haudren, 2018-04-12:
Indeed, those terms are not exposed yet. I've had success using [RBDL](https://rbdl.bitbucket.io/) and [RBDyn](https://github.com/jrl-umi3218/RBDyn) (and maybe even [Tasks](https://github.com/jrl-umi3218/Tasks) if you want to do QP-based control) to compute the missing terms, treating pybullet as a pure simulator.

erwincoumans, 2018-04-12:
It would be good to expose the missing terms in PyBullet as well.
1) Exposing the N term (gravity, coriolis and centrifugal) should be easy.

2) Why would you need the \tao / user force? Didn't you apply it yourself?

3) Do you need the constaint forces, as computed by the LCP solver, such as contact, joint limits term too?

Any other term(s)?

haudren, 2018-04-24:
Sorry, I did not come back to this issue:

RBDL/RBDyn only help with your point 1) AFAIK. However, RBDyn especially has a suite of associated control tools.

2) In my mind, `\tau` is the control signal, so it is provided by the user.

3) In typical control schemes, the contact forces are either sensed or estimated. I think it would be important to be able to retrieve that information from the simulator to implement and / or verify algorithms. An extra term might be the contact locations.

elkmug, 2018-06-04:
>     1. Exposing the N term (gravity, coriolis and centrifugal) should be easy.

If we're voting, heck even if we're not, I'd love to see this exposed in pybullet.

foolyc, 2018-06-04:
Thanks for your reply. I choose RBDL in my project.

benelot, 2018-06-04:
Constraint forces might be something I need at some point for openai gym.

erwincoumans, 2018-11-27:
I'm currently re-implementing DeepMimic in PyBullet and for stable PD control also need the N (or C, gravity) term. You should be able to get the N term using pybullet.calculateInverseDynamics and setting the target acceleration to zero vector:

N(q,qdot) = p.calculateInverseDynamics(objUid, q, qdot, zeroAcc)

The contact forces are available too, after stepSimulation. Use pybullet.getContactPoints, and see the [quickstart guide](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.cb0co8y2vuvc) for the applied force field.

The only remaining thing would be to expose the non-contact constraint forces (motors, limits etc) as well.

elkmug, 2018-11-28:
This is cool.  My interest in the N term was because I wanted to experiment with stable PD control, as opposed to regular PD control, in PyBullet.  And, that interest was due to reading about them in the DeepMimic paper.

So I'm doubly happy.

erwincoumans, 2018-11-28:
I'm currently looking to extend BulletInverseDynamics with spherical joints and add stable PD control, but it is takes some time.

DeepMimic Stable PD control is here:
https://github.com/xbpeng/DeepMimic/blob/master/DeepMimicCore/sim/ImpPDController.cpp#L136

Its inverse dynamics is here:
https://github.com/xbpeng/DeepMimic/blob/master/DeepMimicCore/sim/RBDUtil.cpp

soshishimada, 2019-11-14:
In the presence of contact force from a floor, calculateInverseDynamics() function with zero acceleration will return:

N(q,qdot) - '\lamda' = p.calculateInverseDynamics(objUid, q, qdot, zeroAcc).

meaning, the returned value includes centripetal, Coriolis, gravity and force from a floor.  Am I right?

closed 2021-04-26 by erwincoumans