bulletphysics/bullet3 #1530: Energy Conservation Error - Restitution Example

AlCap23, 2018-01-26:
First of all: I'm really enjoying pybullet! Thank you.

I am currently investigating different physic engines for simulation purposes. As a starting point I've basically taken the restitution.py example and played a little bit with it. 

I came across an interesting effect: For small heights of the ball, the ideal bouncyness works fine and the energy / momentum is conserved. If the initial height is set to 10 m, the ball will lose energy when the simulation advances. This does not vary with the step size, solver iterations or substeps.

The resting height of the ideal bouncing ball is ~4.5 m. 

I found some older post in the forum regarding this issue:

-[Energy-conserving collisions?](https://pybullet.org/Bullet/phpBB3/viewtopic.php?p=&f=&t=4703)
-[Restitution dependend on timestep](https://pybullet.org/Bullet/phpBB3/viewtopic.php?t=11860)
-[How can I increase the physical realism of my simulation](https://pybullet.org/Bullet/phpBB3/viewtopic.php?p=&f=9&t=10695)

Is there any explanation / progress with regards to this topic?
I am using double precision, varied step size, switched on splitImpulse, varied erp values. 

An example of the code is attached.  ( Matplotlib is used in a hacky way to observe the system ).
[image]

Code:

```python
#you can set the restitution (bouncyness) of an object in the URDF file
#or using changeDynamics

import pybullet as p
import pybullet_data as pbd
import numpy as np
import matplotlib.pyplot as plt
import matplotlib.animation as animation
import time

def T(v):
	return 0.5 * 10. * v**2
def U(x):
	return 10. * 9.81 * x-1.

cid = p.connect(p.SHARED_MEMORY)
if (cid<0):
	cid = p.connect(p.DIRECT)

# Add a data path to load plane
p.setAdditionalSearchPath(pbd.getDataPath())

# Load the plane
plane = p.loadURDF("plane.urdf")
p.changeDynamics(plane,-1,restitution=1.)

# Load the ball
sphere = p.loadURDF("../../robots/ball-e/urdf/ball-e.urdf",[0,0,10])
p.changeDynamics(sphere,-1,restitution=1.)

p.setRealTimeSimulation(1)
p.setGravity(0,0,-9.81)
p.setPhysicsEngineParameter()

# Set up the measurements
x = np.zeros((2,500), dtype=float, order='C')
w = np.zeros((2,500))

# Make a simple plot
fig, ax = plt.subplots(3, 1, sharex=True, figsize = (10,10))

while True:
	p.stepSimulation()

	# Get measurements
	# Get the position
	pos,orn = p.getBasePositionAndOrientation(sphere)
	# Get velocity
	vel, rot = p.getBaseVelocity(sphere)

	# Manipulate the array
	x[:,:-1] = x[:, 1:]
	# Write new entries
	x[0,-1] = pos[-1]
	x[1,-1] = vel[-1]

	# Manipulate the array
	w[:,:-1] = w[:, 1:]
	# Write new entries
	w[0,-1] = U(pos[-1])
	w[1,-1] = T(vel[-1])

	# plot
	ax[0].clear(), ax[1].clear(), ax[2].clear()
	ax[0].plot(x[0]), ax[0].plot(np.mean(x[0])*np.ones_like(x[0]))
	ax[1].plot(x[1]), ax[1].plot(np.mean(x[1])*np.ones_like(x[0]))
	ax[2].plot(w[0]+w[1]), ax[2].plot(np.mean(w[0]+w[1])*np.ones_like(x[0]))
	plt.pause(0.001) 
```

erwincoumans, 2018-01-26:
There is some default damping active, you can disable it using 
PyBullet.changeDynamics(obUid, -1, linearDamping=0, angularDamping=0
Also, the splitImpulse option is not implemented for multibody, so the error recovery will impact the energy/restitution.

Also, the ball-e.urdf is missing. Can you use the existing sphere2.urdf or attached a zipped version here?

AlCap23, 2018-01-29:
Thanks for the quick reply. I haven't been in office until now, but with the changing of:

```python
p.changeDynamics(plane,-1,restitution=.97)
p.changeDynamics(sphere,-1,restitution=1., linearDamping = 1e-5, angularDamping = 1e-5)
```
 I have been able to produce a stable energy level for both the "sphere2.urdf" and "ball-e.urdf" models.

closed 2018-02-21 by AlCap23