bulletphysics/bullet3 #1537: Collision detection Error with Half Cheetah

FitMachineLearning, 2018-01-27:
The environment fails to detect that the cheetah's head has crashed onto the ground.
Detected while training my agent

You can see it here
https://youtu.be/2w0oUw-pHN0

erwincoumans, 2018-01-27:
This Half Cheetah is a conversion by Benjamin from the original Roboschool Half Cheetah. 
Does that original environment have any collision check with the ground as termination?
https://github.com/openai/roboschool/tree/master/roboschool
At some stage, I'd like to see a PyBullet conversion from the Mujoco Half Cheetah (and other envs) instead of the harder/modified Roboschool environments.

benelot, 2018-01-29:
I will look into this in case there should be a termination upon hitting
ground.

FitMachineLearning, 2018-01-29:
Thanks a bunch Ben.

One other thing to note is that there seems to be a problem with the efficiency-dependant reward allocation. 

Algorithms agent tend to freeze into some algorithm-independent mediocre state. This does not happen with the same algorithms on different continuous control environments.

So far I have observed this problem on HalfCheetah, Ant, Hopper. 
Hopper has additional termination issues where the episode would suddenly terminate when the agent is in the air .

It might be worthwhile, as Erwin suggests, to port over the Mujoco versions.
Or maybe have a coding contest.

benelot, 2018-01-29:
I am already working on porting them over. It takes some experience to do
it properly, that is why I am not sure if a coding contest could do it.
Thanks for the feedback, it is highly appreciated.

FitMachineLearning, 2018-01-29:
Awesome. 
I am going through your repos and I can see it there. 
Let me know if you are looking for someone to test them out.

benelot, 2018-01-30:
Hi, I am totally interested at some point. Do you have MuJoCo as well? I
need an agent trained on a MuJoCo env and if possible a trace of the
position of one/all limbs at each time. Then I can plot that in a debug
view to see where and possibly why it differs.

erwincoumans, 2018-02-01:
I just tested enjoy_TF_HalfCheetahBulletEnv_v0_2017may.py and added a print. The collision between head and ground is correctly reported. Also, we can train Ant, Cheetah etc just fine. It could be an error in your training algorithm? If not, please provide a repro case of a collision/environment issue.

erwincoumans, 2018-02-01:
[debugGym.zip](https://github.com/bulletphysics/bullet3/files/1686901/debugGym.zip)

here is the script to print 'Terminated' etc.

closed 2018-02-01 by erwincoumans

FitMachineLearning, 2018-02-02:
Thanks.

Thanks a bunch Erwin. Will double check again with Algo.
I was able to train hopper with updated pybullet https://youtu.be/A_JkodPVxjQ

I will run the algorithm on other environments again and report.