bulletphysics/bullet3 #1538: build error using premake4_linux64

consultit, 2018-01-28:
Hi I'm trying to build bullet on GNU/Linux (debian 9.3) using these commands
```
$ ./premake4_linux64 --enable_pybullet gmake
$ cd gmake/
$ make all
```
but I get these errors:
```
...
==== Building App_PhysicsServer_SharedMemory (release64) ====
Linking App_PhysicsServer_SharedMemory
obj/x64/Release/App_PhysicsServer_SharedMemory/b3PluginManager.o: In function `b3PluginManager::loadPlugin(char const*, char const*)':
b3PluginManager.cpp:(.text+0x1060): undefined reference to `dlopen'
b3PluginManager.cpp:(.text+0x14cb): undefined reference to `dlsym'
b3PluginManager.cpp:(.text+0x14df): undefined reference to `dlsym'
b3PluginManager.cpp:(.text+0x14f3): undefined reference to `dlsym'
b3PluginManager.cpp:(.text+0x1507): undefined reference to `dlsym'
b3PluginManager.cpp:(.text+0x151b): undefined reference to `dlsym'
obj/x64/Release/App_PhysicsServer_SharedMemory/b3PluginManager.o:b3PluginManager.cpp:(.text+0x152f): more undefined references to `dlsym' follow
obj/x64/Release/App_PhysicsServer_SharedMemory/b3PluginManager.o: In function `b3PluginManager::loadPlugin(char const*, char const*)':
b3PluginManager.cpp:(.text+0x15d2): undefined reference to `dlclose'
b3PluginManager.cpp:(.text+0x1728): undefined reference to `dlclose'
b3PluginManager.cpp:(.text+0x1c80): undefined reference to `dlclose'
obj/x64/Release/App_PhysicsServer_SharedMemory/b3PluginManager.o: In function `b3PluginManager::unloadPlugin(int)':
b3PluginManager.cpp:(.text+0x2a24): undefined reference to `dlclose'
obj/x64/Release/App_PhysicsServer_SharedMemory/b3PluginManager.o: In function `b3PluginManager::registerStaticLinkedPlugin(char const*, int (*)(b3PluginContext*), void (*)(b3PluginContext*), int (*)(b3PluginContext*, b3PluginArguments const*), int (*)(b3PluginContext*), int (*)(b3PluginContext*), UrdfRenderingInterface* (*)(b3PluginContext*))':
b3PluginManager.cpp:(.text+0x3539): undefined reference to `dlclose'
collect2: error: ld returned 1 exit status
App_PhysicsServer_SharedMemory.make:152: recipe for target '../../bin/App_PhysicsServer_SharedMemory_gmake_x64_release' failed
make[1]: *** [../../bin/App_PhysicsServer_SharedMemory_gmake_x64_release] Error 1
Makefile:40: recipe for target 'App_PhysicsServer_SharedMemory' failed
make: *** [App_PhysicsServer_SharedMemory] Error 2
```

EastskyKang, 2018-02-07:
same here. 

I am using cmake 3.5 

 ```
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftBody.o): In function `btSoftBody::updatePose() [clone .part.103] [clone .constprop.409]':
btSoftBody.cpp:(.text+0x1c77): undefined reference to `btPolarDecomposition::decompose(btMatrix3x3 const&, btMatrix3x3&, btMatrix3x3&) const'
btSoftBody.cpp:(.text+0x2003): undefined reference to `btPolarDecomposition::btPolarDecomposition(float, unsigned int)'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftBody.o): In function `btSoftBody::updateClusters()':
btSoftBody.cpp:(.text+0xc154): undefined reference to `btPolarDecomposition::decompose(btMatrix3x3 const&, btMatrix3x3&, btMatrix3x3&) const'
btSoftBody.cpp:(.text+0xcc33): undefined reference to `btPolarDecomposition::btPolarDecomposition(float, unsigned int)'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftBodyRigidBodyCollisionConfiguration.o): In function `btSoftBodyRigidBodyCollisionConfiguration::~btSoftBodyRigidBodyCollisionConfiguration()':
btSoftBodyRigidBodyCollisionConfiguration.cpp:(.text+0x9d): undefined reference to `btDefaultCollisionConfiguration::~btDefaultCollisionConfiguration()'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftBodyRigidBodyCollisionConfiguration.o): In function `btSoftBodyRigidBodyCollisionConfiguration::btSoftBodyRigidBodyCollisionConfiguration(btDefaultCollisionConstructionInfo const&)':
btSoftBodyRigidBodyCollisionConfiguration.cpp:(.text+0xdb): undefined reference to `btDefaultCollisionConfiguration::btDefaultCollisionConfiguration(btDefaultCollisionConstructionInfo const&)'
btSoftBodyRigidBodyCollisionConfiguration.cpp:(.text+0x26a): undefined reference to `btDefaultCollisionConfiguration::~btDefaultCollisionConfiguration()'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftBodyRigidBodyCollisionConfiguration.o): In function `btSoftBodyRigidBodyCollisionConfiguration::~btSoftBodyRigidBodyCollisionConfiguration()':
btSoftBodyRigidBodyCollisionConfiguration.cpp:(.text+0x92): undefined reference to `btDefaultCollisionConfiguration::~btDefaultCollisionConfiguration()'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftBodyRigidBodyCollisionConfiguration.o): In function `btSoftBodyRigidBodyCollisionConfiguration::getCollisionAlgorithmCreateFunc(int, int)':
btSoftBodyRigidBodyCollisionConfiguration.cpp:(.text+0x2d3): undefined reference to `btDefaultCollisionConfiguration::getCollisionAlgorithmCreateFunc(int, int)'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftBodyRigidBodyCollisionConfiguration.o):(.rodata._ZTI41btSoftBodyRigidBodyCollisionConfiguration[_ZTI41btSoftBodyRigidBodyCollisionConfiguration]+0x10): undefined reference to `typeinfo for btDefaultCollisionConfiguration'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftBodyRigidBodyCollisionConfiguration.o):(.rodata._ZTV41btSoftBodyRigidBodyCollisionConfiguration[_ZTV41btSoftBodyRigidBodyCollisionConfiguration]+0x38): undefined reference to `btDefaultCollisionConfiguration::getClosestPointsAlgorithmCreateFunc(int, int)'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftRigidCollisionAlgorithm.o): In function `btSoftRigidCollisionAlgorithm::btSoftRigidCollisionAlgorithm(btPersistentManifold*, btCollisionAlgorithmConstructionInfo const&, btCollisionObjectWrapper const*, btCollisionObjectWrapper const*, bool)':
btSoftRigidCollisionAlgorithm.cpp:(.text+0xc0): undefined reference to `btCollisionAlgorithm::btCollisionAlgorithm(btCollisionAlgorithmConstructionInfo const&)'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftMultiBodyDynamicsWorld.o): In function `btSoftMultiBodyDynamicsWorld::serialize(btSerializer*)':
btSoftMultiBodyDynamicsWorld.cpp:(.text+0x9ab): undefined reference to `btDiscreteDynamicsWorld::serializeDynamicsWorldInfo(btSerializer*)'
btSoftMultiBodyDynamicsWorld.cpp:(.text+0x9d1): undefined reference to `btDiscreteDynamicsWorld::serializeRigidBodies(btSerializer*)'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftMultiBodyDynamicsWorld.o): In function `btSoftMultiBodyDynamicsWorld::~btSoftMultiBodyDynamicsWorld()':
btSoftMultiBodyDynamicsWorld.cpp:(.text+0xb26): undefined reference to `btMultiBodyDynamicsWorld::~btMultiBodyDynamicsWorld()'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftMultiBodyDynamicsWorld.o): In function `btSoftMultiBodyDynamicsWorld::btSoftMultiBodyDynamicsWorld(btDispatcher*, btBroadphaseInterface*, btMultiBodyConstraintSolver*, btCollisionConfiguration*, btSoftBodySolver*)':
btSoftMultiBodyDynamicsWorld.cpp:(.text+0xb87): undefined reference to `btMultiBodyDynamicsWorld::btMultiBodyDynamicsWorld(btDispatcher*, btBroadphaseInterface*, btMultiBodyConstraintSolver*, btCollisionConfiguration*)'
btSoftMultiBodyDynamicsWorld.cpp:(.text+0x1159): undefined reference to `btMultiBodyDynamicsWorld::~btMultiBodyDynamicsWorld()'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftMultiBodyDynamicsWorld.o): In function `btSoftMultiBodyDynamicsWorld::predictUnconstraintMotion(float)':
btSoftMultiBodyDynamicsWorld.cpp:(.text+0x118f): undefined reference to `btDiscreteDynamicsWorld::predictUnconstraintMotion(float)'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftMultiBodyDynamicsWorld.o): In function `btSoftMultiBodyDynamicsWorld::internalSingleStepSimulation(float)':
btSoftMultiBodyDynamicsWorld.cpp:(.text+0x1253): undefined reference to `btDiscreteDynamicsWorld::internalSingleStepSimulation(float)'
[... cut ...]
examples/SharedMemory/CMakeFiles/App_PhysicsServer_SharedMemory.dir/build.make:1456: recipe for target 'examples/SharedMemory/App_PhysicsServer_SharedMemory' failed
make[2]: *** [examples/SharedMemory/App_PhysicsServer_SharedMemory] Error 1
CMakeFiles/Makefile2:631: recipe for target 'examples/SharedMemory/CMakeFiles/App_PhysicsServer_SharedMemory.dir/all' failed
make[1]: *** [examples/SharedMemory/CMakeFiles/App_PhysicsServer_SharedMemory.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
```

erwincoumans, 2018-02-07:
What if you use pip install pybullet instead (or sudo pip install pybullet or sudo pip3 install pybullet)?

closed 2018-02-07 by erwincoumans

consultit, 2018-02-09:
Hi unfortunately premake at least until commit 0421c60, does not build correctly with gcc 6.3.0 (Debian 9.3) and gcc 5.4.0 (Ubuntu 16.04).
After analyzing a bit I saw that the problems are related to the linking phase:
- the plugin system does not link with the `dl` library
- the linker complains that various static libraries (ie declared `kind StaticLib`) are not compiled with the `-fPIC` flag

To solve the first problem I have included, in several premake.lua, the code:
```
if os.is ("Linux") then
     Links {"dl"}
end
```
The second I solved by inserting in all the premake.lua that define the static libraries the code:
```
     if os.is ("Linux") then
         buildoptions {"- fPIC"}
     end
```
To see the changes made, I've attached a simple `git diff master` with the latest commit 0421c60
[bullet3-premake.diff.txt](https://github.com/bulletphysics/bullet3/files/1710791/bullet3-premake.diff.txt)

Bye

erwincoumans, 2018-02-10:
Can you try latest revision, this pull request should fix it for premake and cmake:
https://github.com/bulletphysics/bullet3/pull/1557