bulletphysics/bullet3 #1543: Query: Is there any api for creating a convex hull with set of points and ask for collision infos

k-maheshkumar, 2018-02-01:
Hi,
I am trying to implement a motion planner from this [paper](http://www.eecs.berkeley.edu/~joschu/docs/trajopt-paper.pdf), where collision information are need so that Jacobian can be used to avoid collision during planning.

Further, in this paper says that the discrete samples of a trajectory is not collision free in continuous time,  so we need to find the closest point between the convex hull made by A(t) and A(t+1) to have a collision free trajectory in continuous time.

So, my query is that, is there any way to form a convexhull between set of points in a trajectory and call getClosestPoint from this created convexhull.

Thanks in advance ..

erwincoumans, 2018-02-01:
You can store a convex point cloud as an .obj or .stl file. Then create a URDF file that points to the obj file. See all the examples in PyBullet and the data folder.

closed 2018-02-01 by erwincoumans

k-maheshkumar, 2018-02-01:
Nope. I don't have point clouds as such. All I have is two joint angles(samples) corresponding to a link of a trajectory. Now I need to find a closest point between the convexhull formed between two successive sample points of that link with a nearby obstacle.

k-maheshkumar, 2018-02-02:
From the documentation, I can see that, method rayTest returns collision information such as hit normal, hit position etc. But, I need same information along with signed(separation or penetration) distance, how we get it from getClosestPoints method around certain distance, incase of no collision.

Is there any workaround to get similar results?

Please help me with this.

erwincoumans, 2018-02-02:
PyBullet.getClosestPoints gives the closest points, normal and signed distance, no workaround needed.