bulletphysics/bullet3 #1545: Slow Rendering in pybullet

BlGene, 2018-02-02:
For RL I would like to have pybullet render as fast as possible. How would I best achieve this?
I tested the three connection modes with ER_BULLET_HARDWARE_OPENGL and got the following fps:

DIRECT +shadow 29 ( defaults to ER_TINY_RENDERER?)
DIRECT -shadow 47 (defaults to ER_TINY_RENDERER?)
SHARED_MEMORY +shadow  29 ( using App_PhysicsServer_SharedMemory)
SHARED_MEMORY -shadow   47 ( using App_PhysicsServer_SharedMemory)
(to test FPS times I used: https://gist.github.com/BlGene/9010fb3671dac3cc566ee659fcda155a)

The GUI or SHARED_MEMORY + App_ExampleBrowser connections run very slowly for me <10fps. This seems to work better others, a youtube video (https://www.youtube.com/watch?v=__ilQzkPDNI)  mentions the ExampleBrowser running at 1000fps. 

The difference may be due to pybullet is selecting the wrong GPU on my machine (NVS 310 vs  GeForce GTX 1070).   Is there a way to control which GPU is used by pybullet?

Training for RL with the GUI connection would require the option not to open the ExampleBrowser window. I expected COV_ENABLE_GUI to do this. Would this be sufficient to allow running in headless mode?

The following line the documentation is confusing for me:
    "Note that DIRECT mode has no OpenGL, so it requires ER_TINY_RENDERER."
From this I understood that App_PhysicsServer_SharedMemory uses OpenGL, although its never explicitly stated. This does however not appear to be the cause as shown by fps performance and GPU memory consumption.

To get fast OpenGL based rendering another option would be to use the recently introduced dynamically loaded render plugins. One approach would be to adapt the roboschool rendering as a plugin. But before I attempt this I would like to ask if it is this a reasonable approach and if this code has already been written by somebody else.

Thanks for any help in advance.

BlGene, 2018-02-02:
Also: 

1. In the pybullet documentation google doc the setJointMotorControlArray input variable linkIndices should be jointIndices.
2. Should a way to access btCollisionShape::setMargin from pybullet be added?

erwincoumans, 2018-02-02:
You cannot easily train in GUI or SHARED_MEMORY mode unless you only have 1 thread. You could just use DIRECT mode and TinyRenderer. Apparently, rendering the ground plane takes most time. TinyRenderer hasn't been optimized.

>> https://www.youtube.com/watch?v=__ilQzkPDNI

Yes, this example runs over 1000FPS, but there is no 'getCameraImage' used: most of the locomotion RL tasks don't use vision/camera.

>> roboschool rendering 

That won't help, Bullet's own OpenGL renderer is likely faster than roboschool's one. Time is currently going into other parts.

When in GUI / hardware OpenGL mode, the pixels are copied from the main window. Resizing it to match the requested camera image dimensions helps. Also disable the GUI (= buttons etc). See attached updated version.
[rendertest.zip](https://github.com/bulletphysics/bullet3/files/1690473/rendertest.zip)

Multi-GPU setups are a hassle indeed, can you just take out the GPU?
See also https://github.com/bulletphysics/bullet3/pull/1481

>>setJointMotorControlArray input variable linkIndices should be jointIndices.

Thanks, fixed

>> btCollisionShape::setMargin 

No, setMargin is not exposed, it is not consistently implemented for various shapes, and would cause confusion.

Later this year we will use vision/synthetic camera and likely optimize the 'getCameraImage' implementation. Note that very little time is currently spend in rendering (less than 200 microseconds on my 1080), it is mainly copying data / waiting that makes is several milliseconds.

When you run my updated script, there is additional performance logging. The renderTimings.json can be opened in Google Chrome using about://tracing.

[image]

erwincoumans, 2018-02-02:
The preview windows in the BulletExampleBrowser (GUI, SHARED_MEMORY etc) takes some time too. You can make it faster (when using renderer=ER_BULLET_HARDWARE_OPENGL) by disabling them:

~~~~~~~~
 optionstring='--width={} --height={}'.format(pixelWidth,pixelHeight)
        
        cid = pybullet.connect(self.connection_mode, options=optionstring,*self.argv)
        if cid < 0:
            raise ValueError
        print("connected")
        pybullet.configureDebugVisualizer(pybullet.COV_ENABLE_GUI,0)
        pybullet.configureDebugVisualizer(pybullet.COV_ENABLE_SEGMENTATION_MARK_PREVIEW,0)
        pybullet.configureDebugVisualizer(pybullet.COV_ENABLE_DEPTH_BUFFER_PREVIEW,0)
        pybullet.configureDebugVisualizer(pybullet.COV_ENABLE_RGB_BUFFER_PREVIEW,0)
~~~~~~~~

BlGene, 2018-02-02:
Hi Erwin,

thanks for the quick reply. From what I understand:

1. DIRECT connection mode does not use GPU rendering.
2. The one thread per bullet instance constraint for GUI / DIRECT is
satisfied if I run bullet instances in separate processes using
`multiprocessing.Process` as done by SubprocVecEnv.
3. What FPS are you getting for getting for BulletExampleBrowser w/
and w/o preview
windows.
4. The fastest solution would probably be to write a TF GPU Op which uses
cuda's OpenGL Interoperability API to do a OpenGL->TF copy. This is a bit
advanced for me, but cursory googling suggest this would require sharing
OpenGL contexts between processes via GLX_EXT_import_context.

Best, Max

erwincoumans, 2018-02-02:
>> 1. DIRECT connection mode does not use GPU rendering.

Indeed. DIRECT  means each PyBullet call is directly process in the same thread, using purely CPU (=TinyRenderer at the moment).

>> 2. The one thread per bullet instance constraint for GUI / DIRECT is
satisfied if I run bullet instances in separate processes using
`multiprocessing.Process` as done by SubprocVecEnv.

Yes, if you use separate processes, you can use multiple examplebrowsers / GUI mode.

>> 3.  What FPS are you getting for getting for BulletExampleBrowser w/
and w/o preview windows. (edit: this was answered, thanks)

Around 280 FPS, which is pretty good.

>> 4. The fastest solution would probably be to write a TF GPU Op which uses
cuda's OpenGL Interoperability API to do a OpenGL->TF copy. This is a bit
advanced for me, but cursory googling suggest this would require sharing
OpenGL contexts between processes via GLX_EXT_import_context.

Yes, this would require a lot of effort.

BlGene, 2018-02-02:
So in your last post on the image "Rendering" -> getCameraImage and "Render2" ->BulletExampleBrowser.

4ms? there's only like .4ms of real rendering in there. How do I parallelize rendering/physics? :rocket:  :grin:

erwincoumans, 2018-02-02:
Things are already parallelized, physics runs in a separate thread from graphics.

Rendering + readback GPU is around 2.5 ms. I wouldn't waste time going from 4ms to 2.5ms, hence take 280FPS as your target.

BlGene, 2018-02-02:
Ok, I'll try to add a method for GUI not to open an examplebrowser window. This would allow training.

erwincoumans, 2018-02-02:
There is already some implementation using EGLOpenGLWindow. We used this internally in Google in the cloud. It will open a context using EGL (not X11) and the window is not visible. What is your target platform?

BlGene, 2018-02-02:
Ubuntu with Nvidia Titian X, I think EGLOpenGLWindow should work for me.

BlGene, 2018-02-02:
Is this implementation based on the GUI connection mode (examplebrowser without examplebrowser) or SHARED_MEMORY (App_PhysicsServer_SharedMemory).

If it would be possible to make this option available on github that would be very helpful for me and others too.

erwincoumans, 2018-02-02:
EGLOpenGLWindow is based on GUI mode. No need to 'reduce' anything.

erwincoumans, 2018-02-02:
You need to define BT_USE_EGL in your build system, and make it all compile again. I don't have time to dig into this right now.

BlGene, 2018-02-02:
👍 thanks a lot  for all of the help!

I’ll try this next week.

BlGene, 2018-02-16:
I have tried to create a PR to make this stuff a bit easier, please see: #1565.

closed 2018-02-28 by BlGene

Rkartik27, 2019-06-12:
Hi Erwin,

I am using a Nvidia Quadro M5000M GPU (8GB) in Ubuntu 16.04 with python 3.5.2 version. I am having a problem in replicating such a high FPS (you said that you are getting a 280 FPS) with my computer. I have even tried building pybullet from the source code using "python3 setup.py install --user" from the latest source code as I heard that source built pybullet might have GPU support.

[image]

pybullet also seems to be using the GPU in my laptop for the code shared by you in rendertest.zip (link: https://github.com/bulletphysics/bullet3/files/1690473/rendertest.zip). The screenshot I shared above shows my nvidia-smi output in which python3 is the memory being used by the pybullet code (please ignore the other python 2 processes; other Tensorflow codes are using the GPU). However I am only getting an average **FPS of around 20** as seen in the print output below:

pybullet build time: Jun 12 2019 17:39:14
connecting
argv[0]=--width=640
argv[1]=--height=480
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=4
argv[0] = --unused
argv[1] = --width=640
argv[2] = --height=480
argv[3] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.3 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=NVIDIA Corporation
GL_RENDERER=Quadro M5000M/PCIe/SSE2
GL_VERSION=3.3.0 NVIDIA 396.54
GL_SHADING_LANGUAGE_VERSION=3.30 NVIDIA via Cg compiler
pthread_getconcurrency()=0
Version = 3.3.0 NVIDIA 396.54
Vendor = NVIDIA Corporation
Renderer = Quadro M5000M/PCIe/SSE2
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
connected
ven = NVIDIA Corporation
Testing SHARED
mean: 19.56697039069384 for 100 runs

Testing SHARED w/o shadow
mean: 18.793571272315447 for 100 runs

Writing 17186 timings for thread 0
Writing 2202 timings for thread 1
Writing 47693 timings for thread 2
numActiveThreads = 0
stopping threads
Thread with taskId 0 exiting
destroy semaphore
semaphore destroyed
Thread TERMINATED
destroy main semaphore
main semaphore destroyed
finished
numActiveThreads = 0
btShutDownExampleBrowser stopping threads
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed

Any help in my case would be highly appreciated, thx

baoyufuyou, 2020-06-23:
Hi guys, I faced also problem of low rendering time. I tested the speed in my computer. Then, the getscreenpixels20423 will take about ```17 ms```. 
with GPU 0: ```GeForce RTX 2070```. Is that normal????? I'm a little bit confused.
Thanks a lot. Guys.
[image]

[image]
[image]

BlGene, 2020-06-24:
@baoyufuyou 
Hi Lei,
I have two tips that you could try.
  1. (Just to double check) have a look at the program output, it should show something like ```Loaded EGL 1.5 after reload```
  2. When calling `pybullet.connect`, also setting the `width` and `height` variables, see snippet below.

```
       if self._renderer == "tiny":
            optionstring = '--width={} --height={}'.format(
                render_width, render_height)
            cid = p.connect(p.DIRECT, options=optionstring)
        elif self._renderer == "egl":
            optionstring = '--width={} --height={}'.format(
                render_width, render_height)
            # optional --window_backend and --render_device, not implemented
            # yet
            cid = p.connect(p.DIRECT, options=optionstring)
```

baoyufuyou, 2020-07-03:
@BlGene Thank you very much. ;)