bulletphysics/bullet3 #1554: build error with cmake

EastskyKang, 2018-02-07:
I am using cmake 3.5.

I got the error as follows as I tried to build Bullet with Cmake. 

```sh
$ mkdir build  # at bullet root 
$ cmake ../
$ make -j
$ sudo make install
```

```
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftBody.o): In function `btSoftBody::updatePose() [clone .part.103] [clone .constprop.409]':
btSoftBody.cpp:(.text+0x1c77): undefined reference to `btPolarDecomposition::decompose(btMatrix3x3 const&, btMatrix3x3&, btMatrix3x3&) const'
btSoftBody.cpp:(.text+0x2003): undefined reference to `btPolarDecomposition::btPolarDecomposition(float, unsigned int)'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftBody.o): In function `btSoftBody::updateClusters()':
btSoftBody.cpp:(.text+0xc154): undefined reference to `btPolarDecomposition::decompose(btMatrix3x3 const&, btMatrix3x3&, btMatrix3x3&) const'
btSoftBody.cpp:(.text+0xcc33): undefined reference to `btPolarDecomposition::btPolarDecomposition(float, unsigned int)'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftBodyRigidBodyCollisionConfiguration.o): In function `btSoftBodyRigidBodyCollisionConfiguration::~btSoftBodyRigidBodyCollisionConfiguration()':
btSoftBodyRigidBodyCollisionConfiguration.cpp:(.text+0x9d): undefined reference to `btDefaultCollisionConfiguration::~btDefaultCollisionConfiguration()'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftBodyRigidBodyCollisionConfiguration.o): In function `btSoftBodyRigidBodyCollisionConfiguration::btSoftBodyRigidBodyCollisionConfiguration(btDefaultCollisionConstructionInfo const&)':
btSoftBodyRigidBodyCollisionConfiguration.cpp:(.text+0xdb): undefined reference to `btDefaultCollisionConfiguration::btDefaultCollisionConfiguration(btDefaultCollisionConstructionInfo const&)'
btSoftBodyRigidBodyCollisionConfiguration.cpp:(.text+0x26a): undefined reference to `btDefaultCollisionConfiguration::~btDefaultCollisionConfiguration()'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftBodyRigidBodyCollisionConfiguration.o): In function `btSoftBodyRigidBodyCollisionConfiguration::~btSoftBodyRigidBodyCollisionConfiguration()':
btSoftBodyRigidBodyCollisionConfiguration.cpp:(.text+0x92): undefined reference to `btDefaultCollisionConfiguration::~btDefaultCollisionConfiguration()'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftBodyRigidBodyCollisionConfiguration.o): In function `btSoftBodyRigidBodyCollisionConfiguration::getCollisionAlgorithmCreateFunc(int, int)':
btSoftBodyRigidBodyCollisionConfiguration.cpp:(.text+0x2d3): undefined reference to `btDefaultCollisionConfiguration::getCollisionAlgorithmCreateFunc(int, int)'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftBodyRigidBodyCollisionConfiguration.o):(.rodata._ZTI41btSoftBodyRigidBodyCollisionConfiguration[_ZTI41btSoftBodyRigidBodyCollisionConfiguration]+0x10): undefined reference to `typeinfo for btDefaultCollisionConfiguration'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftBodyRigidBodyCollisionConfiguration.o):(.rodata._ZTV41btSoftBodyRigidBodyCollisionConfiguration[_ZTV41btSoftBodyRigidBodyCollisionConfiguration]+0x38): undefined reference to `btDefaultCollisionConfiguration::getClosestPointsAlgorithmCreateFunc(int, int)'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftRigidCollisionAlgorithm.o): In function `btSoftRigidCollisionAlgorithm::btSoftRigidCollisionAlgorithm(btPersistentManifold*, btCollisionAlgorithmConstructionInfo const&, btCollisionObjectWrapper const*, btCollisionObjectWrapper const*, bool)':
btSoftRigidCollisionAlgorithm.cpp:(.text+0xc0): undefined reference to `btCollisionAlgorithm::btCollisionAlgorithm(btCollisionAlgorithmConstructionInfo const&)'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftMultiBodyDynamicsWorld.o): In function `btSoftMultiBodyDynamicsWorld::serialize(btSerializer*)':
btSoftMultiBodyDynamicsWorld.cpp:(.text+0x9ab): undefined reference to `btDiscreteDynamicsWorld::serializeDynamicsWorldInfo(btSerializer*)'
btSoftMultiBodyDynamicsWorld.cpp:(.text+0x9d1): undefined reference to `btDiscreteDynamicsWorld::serializeRigidBodies(btSerializer*)'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftMultiBodyDynamicsWorld.o): In function `btSoftMultiBodyDynamicsWorld::~btSoftMultiBodyDynamicsWorld()':
btSoftMultiBodyDynamicsWorld.cpp:(.text+0xb26): undefined reference to `btMultiBodyDynamicsWorld::~btMultiBodyDynamicsWorld()'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftMultiBodyDynamicsWorld.o): In function `btSoftMultiBodyDynamicsWorld::btSoftMultiBodyDynamicsWorld(btDispatcher*, btBroadphaseInterface*, btMultiBodyConstraintSolver*, btCollisionConfiguration*, btSoftBodySolver*)':
btSoftMultiBodyDynamicsWorld.cpp:(.text+0xb87): undefined reference to `btMultiBodyDynamicsWorld::btMultiBodyDynamicsWorld(btDispatcher*, btBroadphaseInterface*, btMultiBodyConstraintSolver*, btCollisionConfiguration*)'
btSoftMultiBodyDynamicsWorld.cpp:(.text+0x1159): undefined reference to `btMultiBodyDynamicsWorld::~btMultiBodyDynamicsWorld()'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftMultiBodyDynamicsWorld.o): In function `btSoftMultiBodyDynamicsWorld::predictUnconstraintMotion(float)':
btSoftMultiBodyDynamicsWorld.cpp:(.text+0x118f): undefined reference to `btDiscreteDynamicsWorld::predictUnconstraintMotion(float)'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftMultiBodyDynamicsWorld.o): In function `btSoftMultiBodyDynamicsWorld::internalSingleStepSimulation(float)':
btSoftMultiBodyDynamicsWorld.cpp:(.text+0x1253): undefined reference to `btDiscreteDynamicsWorld::internalSingleStepSimulation(float)'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftMultiBodyDynamicsWorld.o): In function `btSoftMultiBodyDynamicsWorld::debugDrawWorld()':
btSoftMultiBodyDynamicsWorld.cpp:(.text+0x156e): undefined reference to `btDiscreteDynamicsWorld::debugDrawWorld()'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftMultiBodyDynamicsWorld.o): In function `btSoftMultiBodyDynamicsWorld::~btSoftMultiBodyDynamicsWorld()':
btSoftMultiBodyDynamicsWorld.cpp:(.text+0xae5): undefined reference to `btMultiBodyDynamicsWorld::~btMultiBodyDynamicsWorld()'
btSoftMultiBodyDynamicsWorld.cpp:(.text+0xb0f): undefined reference to `btMultiBodyDynamicsWorld::~btMultiBodyDynamicsWorld()'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftMultiBodyDynamicsWorld.o): In function `btSoftMultiBodyDynamicsWorld::removeCollisionObject(btCollisionObject*)':
btSoftMultiBodyDynamicsWorld.cpp:(.text+0x14da): undefined reference to `btDiscreteDynamicsWorld::removeCollisionObject(btCollisionObject*)'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftMultiBodyDynamicsWorld.o): In function `btDiscreteDynamicsWorld::updateVehicles(float)':
btSoftMultiBodyDynamicsWorld.cpp:(.text._ZN23btDiscreteDynamicsWorld14updateVehiclesEf[_ZN23btDiscreteDynamicsWorld14updateVehiclesEf]+0x1): undefined reference to `btDiscreteDynamicsWorld::updateActions(float)'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftMultiBodyDynamicsWorld.o):(.rodata._ZTI28btSoftMultiBodyDynamicsWorld[_ZTI28btSoftMultiBodyDynamicsWorld]+0x10): undefined reference to `typeinfo for btMultiBodyDynamicsWorld'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftMultiBodyDynamicsWorld.o):(.rodata._ZTV28btSoftMultiBodyDynamicsWorld[_ZTV28btSoftMultiBodyDynamicsWorld]+0x58): undefined reference to `btDiscreteDynamicsWorld::addCollisionObject(btCollisionObject*, int, int)'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftMultiBodyDynamicsWorld.o):(.rodata._ZTV28btSoftMultiBodyDynamicsWorld[_ZTV28btSoftMultiBodyDynamicsWorld]+0x78): undefined reference to `btDiscreteDynamicsWorld::stepSimulation(float, int, float)'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftMultiBodyDynamicsWorld.o):(.rodata._ZTV28btSoftMultiBodyDynamicsWorld[_ZTV28btSoftMultiBodyDynamicsWorld]+0x80): undefined reference to `btDiscreteDynamicsWorld::addConstraint(btTypedConstraint*, bool)'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftMultiBodyDynamicsWorld.o):(.rodata._ZTV28btSoftMultiBodyDynamicsWorld[_ZTV28btSoftMultiBodyDynamicsWorld]+0x88): undefined reference to `btDiscreteDynamicsWorld::removeConstraint(btTypedConstraint*)'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftMultiBodyDynamicsWorld.o):(.rodata._ZTV28btSoftMultiBodyDynamicsWorld[_ZTV28btSoftMultiBodyDynamicsWorld]+0x90): undefined reference to `btDiscreteDynamicsWorld::addAction(btActionInterface*)'
[... cut ...]
examples/SharedMemory/CMakeFiles/App_PhysicsServer_SharedMemory.dir/build.make:1456: recipe for target 'examples/SharedMemory/App_PhysicsServer_SharedMemory' failed
make[2]: *** [examples/SharedMemory/App_PhysicsServer_SharedMemory] Error 1
CMakeFiles/Makefile2:631: recipe for target 'examples/SharedMemory/CMakeFiles/App_PhysicsServer_SharedMemory.dir/all' failed
make[1]: *** [examples/SharedMemory/CMakeFiles/App_PhysicsServer_SharedMemory.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
```

erwincoumans, 2018-02-07:
I just ran exactly those lines, and there is no problem. More info is needed:

~~~~~~~~
[  1%] Built target LinearMath
[ 11%] Built target BulletCollision
[ 16%] Built target BulletDynamics
[ 16%] Built target App_HelloWorld
[ 17%] Built target App_BasicExample
[ 17%] Built target Bullet3Common
[ 19%] Built target OpenGLWindow
[ 19%] Built target AppBasicExampleGui
[ 20%] Built target BulletExampleBrowserLib
[ 27%] Built target gwen
[ 28%] Built target BussIK
[ 29%] Built target BulletInverseDynamicsUtils
[ 30%] Built target BulletInverseDynamics
[ 31%] Built target BulletSoftBody
[ 45%] Built target App_ExampleBrowser
[ 53%] Built target BulletRobotics
[ 53%] Built target BulletFileLoader
[ 53%] Built target BulletWorldImporter
[ 54%] Built target App_RobotSimulator_NoGUI
[ 54%] Built target App_RobotSimulator
[ 61%] Built target App_PhysicsServer_SharedMemory_VR
[ 67%] Built target App_PhysicsServer_SharedMemory_GUI
[ 73%] Built target App_PhysicsServer_SharedMemory
[ 73%] Built target clsocket
[ 80%] Built target App_TwoJoint
[ 80%] Built target App_obj2sdf
[ 82%] Built target ConvexDecomposition
[ 83%] Built target HACD
[ 83%] Built target GIMPACTUtils
[ 84%] Built target BulletXmlWorldImporter
[ 87%] Built target Bullet3OpenCL_clew
[ 88%] Built target Bullet2FileLoader
[ 89%] Built target Bullet3Dynamics
[ 90%] Built target Bullet3Collision
[ 90%] Built target Bullet3Geometry
[ 91%] Built target gtest
[ 91%] Built target Test_BulletInverseDynamicsJacobian
[ 94%] Built target Test_BulletInverseForwardDynamics
[ 94%] Built target Test_BulletInverseDynamics
[ 99%] Built target Test_PhysicsClientServer
[100%] Built target Test_Collision
[100%] Built target Test_btKinematicCharacterController
[100%] Built target Test_BulletDynamics
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/lib/pkgconfig/bullet.pc
-- Installing: /usr/local/lib/cmake/bullet/UseBullet.cmake
-- Installing: /usr/local/lib/cmake/bullet/BulletConfig.cmake
-- Installing: /usr/local/lib/libclsocket.a
-- Installing: /usr/local/include/ActiveSocket.h
-- Installing: /usr/local/include/Host.h
-- Installing: /usr/local/include/PassiveSocket.h
-- Installing: /usr/local/include/SimpleSocket.h
-- Installing: /usr/local/include/StatTimer.h
-- Installing: /usr/local/lib/libBulletInverseDynamicsUtils.a
-- Installing: /usr/local/include/bullet/InverseDynamics
-- Installing: /usr/local/include/bullet/InverseDynamics/BulletInverseDynamicsUtilsCommon.h
-- Installing: /usr/local/include/bullet/PhysicsClientC_API.h
-- Installing: /usr/local/include/bullet/PhysicsClientSharedMemory_C_API.h
-- Installing: /usr/local/include/bullet/PhysicsClientSharedMemory2_C_API.h
-- Installing: /usr/local/include/bullet/PhysicsDirectC_API.h
-- Installing: /usr/local/include/bullet/PhysicsClientUDP_C_API.h
-- Installing: /usr/local/include/bullet/PhysicsClientTCP_C_API.h
-- Installing: /usr/local/include/bullet/SharedMemoryInProcessPhysicsC_API.h
-- Installing: /usr/local/include/bullet/SharedMemoryPublic.h
-- Installing: /usr/local/lib/libBulletRobotics.a
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ranlib: file: /usr/local/lib/libBulletRobotics.a(Win32SharedMemory.o) has no symbols
[... 469 lines cut ...]
-- Installing: /usr/local/include/bullet/Bullet3Common/b3StackAlloc.h
-- Installing: /usr/local/include/bullet/Bullet3Common/b3TransformUtil.h
-- Installing: /usr/local/include/bullet/Bullet3Common/b3Random.h
-- Installing: /usr/local/include/bullet/Bullet3Common/b3Transform.h
-- Installing: /usr/local/include/bullet/Bullet3Common/b3AlignedObjectArray.h
-- Installing: /usr/local/include/bullet/Bullet3Common/b3PoolAllocator.h
-- Installing: /usr/local/include/bullet/Bullet3Common/shared
-- Installing: /usr/local/include/bullet/Bullet3Common/shared/b3Int4.h
-- Installing: /usr/local/include/bullet/Bullet3Common/shared/b3Mat3x3.h
-- Installing: /usr/local/include/bullet/Bullet3Common/shared/b3Float4.h
-- Installing: /usr/local/include/bullet/Bullet3Common/shared/b3Int2.h
-- Installing: /usr/local/include/bullet/Bullet3Common/shared/b3Quat.h
-- Installing: /usr/local/include/bullet/Bullet3Common/shared/b3PlatformDefinitions.h
-- Installing: /usr/local/include/bullet/Bullet3Common/b3Quaternion.h
-- Installing: /usr/local/include/bullet/Bullet3Common/b3MinMax.h
-- Installing: /usr/local/include/bullet/Bullet3Common/b3Matrix3x3.h
-- Installing: /usr/local/include/bullet/Bullet3Common/b3Logging.h
-- Installing: /usr/local/include/bullet/Bullet3Common/b3HashMap.h
-- Installing: /usr/local/include/bullet/Bullet3Common/b3Scalar.h
-- Installing: /usr/local/include/bullet/Bullet3Common/b3CommandLineArgs.h
-- Installing: /usr/local/include/bullet/Bullet3Common/b3FileUtils.h
-- Installing: /usr/local/include/bullet/Bullet3Common/b3AlignedAllocator.h
-- Installing: /usr/local/include/bullet/Bullet3Common/b3QuadWord.h
-- Installing: /usr/local/include/bullet/Bullet3Common/b3ResizablePool.h
~~~~~~~~

closed 2018-02-07 by erwincoumans

EastskyKang, 2018-02-07:
@erwincoumans  thank you for your response. 

I got the error for linking shared memory

[100%] Linking CXX executable App_PhysicsServer_SharedMemory
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftBody.o): In function `btSoftBody::updatePose() [clone .part.103] [clone .constprop.409]':
btSoftBody.cpp:(.text+0x1c77): undefined reference to `btPolarDecomposition::decompose(btMatrix3x3 const&, btMatrix3x3&, btMatrix3x3&) const'
btSoftBody.cpp:(.text+0x2003): undefined reference to `btPolarDecomposition::btPolarDecomposition(float, unsigned int)'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftBody.o): In function `btSoftBody::updateClusters()':
btSoftBody.cpp:(.text+0xc154): undefined reference to `btPolarDecomposition::decompose(btMatrix3x3 const&, btMatrix3x3&, btMatrix3x3&) const'
btSoftBody.cpp:(.text+0xcc33): undefined reference to `btPolarDecomposition::btPolarDecomposition(float, unsigned int)'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftBodyRigidBodyCollisionConfiguration.o): In function `btSoftBodyRigidBodyCollisionConfiguration::~btSoftBodyRigidBodyCollisionConfiguration()':
btSoftBodyRigidBodyCollisionConfiguration.cpp:(.text+0x9d): undefined reference to `btDefaultCollisionConfiguration::~btDefaultCollisionConfiguration()'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftBodyRigidBodyCollisionConfiguration.o): In function `btSoftBodyRigidBodyCollisionConfiguration::btSoftBodyRigidBodyCollisionConfiguration(btDefaultCollisionConstructionInfo const&)':
btSoftBodyRigidBodyCollisionConfiguration.cpp:(.text+0xdb): undefined reference to `btDefaultCollisionConfiguration::btDefaultCollisionConfiguration(btDefaultCollisionConstructionInfo const&)'
btSoftBodyRigidBodyCollisionConfiguration.cpp:(.text+0x26a): undefined reference to `btDefaultCollisionConfiguration::~btDefaultCollisionConfiguration()'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftBodyRigidBodyCollisionConfiguration.o): In function `btSoftBodyRigidBodyCollisionConfiguration::~btSoftBodyRigidBodyCollisionConfiguration()':
btSoftBodyRigidBodyCollisionConfiguration.cpp:(.text+0x92): undefined reference to `btDefaultCollisionConfiguration::~btDefaultCollisionConfiguration()'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftBodyRigidBodyCollisionConfiguration.o): In function `btSoftBodyRigidBodyCollisionConfiguration::getCollisionAlgorithmCreateFunc(int, int)':
btSoftBodyRigidBodyCollisionConfiguration.cpp:(.text+0x2d3): undefined reference to `btDefaultCollisionConfiguration::getCollisionAlgorithmCreateFunc(int, int)'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftBodyRigidBodyCollisionConfiguration.o):(.rodata._ZTI41btSoftBodyRigidBodyCollisionConfiguration[_ZTI41btSoftBodyRigidBodyCollisionConfiguration]+0x10): undefined reference to `typeinfo for btDefaultCollisionConfiguration'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftBodyRigidBodyCollisionConfiguration.o):(.rodata._ZTV41btSoftBodyRigidBodyCollisionConfiguration[_ZTV41btSoftBodyRigidBodyCollisionConfiguration]+0x38): undefined reference to `btDefaultCollisionConfiguration::getClosestPointsAlgorithmCreateFunc(int, int)'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftRigidCollisionAlgorithm.o): In function `btSoftRigidCollisionAlgorithm::btSoftRigidCollisionAlgorithm(btPersistentManifold*, btCollisionAlgorithmConstructionInfo const&, btCollisionObjectWrapper const*, btCollisionObjectWrapper const*, bool)':
btSoftRigidCollisionAlgorithm.cpp:(.text+0xc0): undefined reference to `btCollisionAlgorithm::btCollisionAlgorithm(btCollisionAlgorithmConstructionInfo const&)'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftMultiBodyDynamicsWorld.o): In function `btSoftMultiBodyDynamicsWorld::serialize(btSerializer*)':
btSoftMultiBodyDynamicsWorld.cpp:(.text+0x9ab): undefined reference to `btDiscreteDynamicsWorld::serializeDynamicsWorldInfo(btSerializer*)'
btSoftMultiBodyDynamicsWorld.cpp:(.text+0x9d1): undefined reference to `btDiscreteDynamicsWorld::serializeRigidBodies(btSerializer*)'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftMultiBodyDynamicsWorld.o): In function `btSoftMultiBodyDynamicsWorld::~btSoftMultiBodyDynamicsWorld()':
btSoftMultiBodyDynamicsWorld.cpp:(.text+0xb26): undefined reference to `btMultiBodyDynamicsWorld::~btMultiBodyDynamicsWorld()'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftMultiBodyDynamicsWorld.o): In function `btSoftMultiBodyDynamicsWorld::btSoftMultiBodyDynamicsWorld(btDispatcher*, btBroadphaseInterface*, btMultiBodyConstraintSolver*, btCollisionConfiguration*, btSoftBodySolver*)':
btSoftMultiBodyDynamicsWorld.cpp:(.text+0xb87): undefined reference to `btMultiBodyDynamicsWorld::btMultiBodyDynamicsWorld(btDispatcher*, btBroadphaseInterface*, btMultiBodyConstraintSolver*, btCollisionConfiguration*)'
btSoftMultiBodyDynamicsWorld.cpp:(.text+0x1159): undefined reference to `btMultiBodyDynamicsWorld::~btMultiBodyDynamicsWorld()'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftMultiBodyDynamicsWorld.o): In function `btSoftMultiBodyDynamicsWorld::predictUnconstraintMotion(float)':
btSoftMultiBodyDynamicsWorld.cpp:(.text+0x118f): undefined reference to `btDiscreteDynamicsWorld::predictUnconstraintMotion(float)'
../../src/BulletSoftBody/libBulletSoftBody.a(btSoftMultiBodyDynamicsWorld.o): In function `btSoftMultiBodyDynamicsWorld::internalSingleStepSimulation(float)':
btSoftMultiBodyDynamicsWorld.cpp:(.text+0x1253): undefined reference to `btDiscreteDynamicsWorld::internalSingleStepSimulation(float)'
[... cut ...]
examples/SharedMemory/CMakeFiles/App_PhysicsServer_SharedMemory.dir/build.make:1456: recipe for target 'examples/SharedMemory/App_PhysicsServer_SharedMemory' failed
make[2]: *** [examples/SharedMemory/App_PhysicsServer_SharedMemory] Error 1
CMakeFiles/Makefile2:631: recipe for target 'examples/SharedMemory/CMakeFiles/App_PhysicsServer_SharedMemory.dir/all' failed
make[1]: *** [examples/SharedMemory/CMakeFiles/App_PhysicsServer_SharedMemory.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

erwincoumans, 2018-02-07:
I still don't have an error, and the continuous integration neither (it runs on windows, linux and mac). Are you using the latest github version of Bullet?

erwincoumans, 2018-02-10:
Can you try latest revision, this pull request should fix it for premake and cmake:
https://github.com/bulletphysics/bullet3/pull/1557

EastskyKang, 2018-02-11:
@erwincoumans great! it works now. Thank you!