bulletphysics/bullet3 #1558: example/HelloWorld works in source, but not with make install. Segfault 11

SlightlyCyborg, 2018-02-11:
I am on MacOs Sierra.

I compiled bullet using the sh script provided. If I go into the `cmake_build` and run the example binaries found there, the HelloWorld example works well.

I then ran a make install which put the includes in `/usr/local/include/bullet` and the dylibs in `/usr/local/lib`

I make a new directory for the hello world example and use the following CMakeLists.txt

`# HelloWorld is a minimal sample creating, stepping and deleting a Bullet dynamics world

INCLUDE_DIRECTORIES(
/usr/local/include/bullet
)

set(CMAKE_BUILD_TYPE Debug)

LINK_LIBRARIES(
 BulletDynamics BulletCollision LinearMath
)

	ADD_EXECUTABLE(App_HelloWorld HelloWorld.cpp)

IF (INTERNAL_ADD_POSTFIX_EXECUTABLE_NAMES)
			SET_TARGET_PROPERTIES(App_HelloWorld PROPERTIES  DEBUG_POSTFIX "_Debug")
			SET_TARGET_PROPERTIES(App_HelloWorld PROPERTIES  MINSIZEREL_POSTFIX "_MinsizeRel")
			SET_TARGET_PROPERTIES(App_HelloWorld PROPERTIES  RELWITHDEBINFO_POSTFIX "_RelWithDebugInfo")
ENDIF(INTERNAL_ADD_POSTFIX_EXECUTABLE_NAMES)`

Simple enough. It compiles. However, when I run the app it makes, I get a segfault 11

Using lldb, I get the following message:

`Process 39402 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=EXC_I386_GPFLT)
    frame #0: 0x00000001000066ca App_HelloWorld`btDefaultMotionState::getWorldTransform(btTransform&) const [inlined] btMatrix3x3::operator=(this=0x0000000100500cf8, m=0x00007ffeefbfe390) at btMatrix3x3.h:104
   101 		// Assignment Operator
   102 		SIMD_FORCE_INLINE btMatrix3x3& operator=(const btMatrix3x3& m)
   103 		{
-> 104 			m_el[0].mVec128 = m.m_el[0].mVec128;
   105 			m_el[1].mVec128 = m.m_el[1].mVec128;
   106 			m_el[2].mVec128 = m.m_el[2].mVec128;
   107
Target 0: (App_HelloWorld) stopped.`

dawsonc, 2019-11-10:
I'm having exactly this issue. I can build the examples within the main bullet directory, but I cannot build any stand-alone files using the bullet library. Is there a preferred way for building stand-alone programs that I'm missing (I'm using cmake)?

dawsonc, 2019-11-10:
> I'm having exactly this issue. I can build the examples within the main bullet directory, but I cannot build any stand-alone files using the bullet library. Is there a preferred way for building stand-alone programs that I'm missing (I'm using cmake)?

My apologies; the issue was a conflicting installation of bullet (from apt-get, as a ROS dependency). Modifying my `CMakeLists.txt` to explicitly include and link the directory for the `vcpkg` installation of bullet solved the issue for me.

closed 2021-04-26 by erwincoumans