bulletphysics/bullet3 #1562: Issue with TORQUE_CONTROL for the base link

mjm522, 2018-02-15:
Hi,

[doubt.zip](https://github.com/bulletphysics/bullet3/files/1729781/doubt.zip)

I was trying  a simple planar arm (finger.urdf attached in the zip) with the following code (doubt2.py in the zip) in torque mode. 

When I try to apply a torque of [4.8, 0., 0, 0] on the arm, the arm does not have any changes to its joint state. If I increase the torque value to [480, 0., 0., 0] the base joint of the arm suddenly moves. But a small torque like [0, 4.8, 0, 0] in the second joint moves the arm with no issues. Other joints also has no issues.

Is it something wrong with the URDF or with this torque control? Since all joints are the same and all inertia of the links are quite small, I am stuck in figuring out what is wrong here.  

Could any one please tell me what is going wrong?

Thanks for the time.

erwincoumans, 2018-02-16:
>> Could any one please tell me what is going wrong?

Please checkout the Bullet Quickstart Guide.

Each motor has a default velocity control mode to keep the velocity zero. You can switch this off, by iterating over all motors and using
~~~~~~~
maxForce = 0
mode = p.VELOCITY_CONTROL
p.setJointMotorControl2(objUid, jointIndex, 	controlMode=mode, force=maxForce)
~~~~~~~

closed 2018-02-16 by erwincoumans

erwincoumans, 2018-02-16:
Also, please let us know what you try to achieve with PyBullet, we can give some hints towards a solution.

mjm522, 2018-02-16:
I am trying to build finger gait planner with bullet. It involves several fingers to trying to manipulate  an object. I was able to achieve a rotation of an object in position mode. Bullet seems to works well (quite stable) with contacts. I was trying to re implement the same using torque mode and faced the issue.