bulletphysics/bullet3 #1567: Spring2Constraint gets the dynamics wrong!

trashpirate, 2018-02-17:
Hi! 

First of all, thanks for an awesome physics engine! I've been using Bullet for my research and noticed strange behavior of the btGeneric6Dof**Spring2**Constraint, which seemed to ignore an important principle of physics. I investigated this with a simple horizontal pendulum simulation: one dof (rotation about vertical axis, spring enabled, no gravity). I did several simulations with different pendulum length (distance from pivot point to mass). To my surprise, the frequency of oscillation did not change, no matter the length of the pendulum -> this is dynamically incorrect! The frequency should be inversely correlated with pendulum length. 
I simulated the same thing with the "depricated" btGeneric6Dof**Spring**Constraint and all of a sudden, without making any other changes, I could see the correct behavior: the frequency increased with decreasing pendulum length!

[image]

If there are any parameters I don't know of that would make Spring2Constraint to behave correctly, I would love to know! Otherwise, it would be great if you could fix the issue in the Spring2Constraint since it effects the accuracy of the physics simulations significantly. To my knowledge, the issue only occurs when the spring is enabled - the frequency of a pendulum driven only by gravity changes with the length.

Thanks!

erwincoumans, 2018-02-17:
I haven't never used or looked into the spring parts of the btGeneric6DofSpring2Constraint (for our robotics / deep reinforcement learning work, we use btMultiBody, and we don't use springs).

Can you please create a small reproduction case, perhaps modifying some Bullet example, that  shows the issue?

trashpirate, 2018-02-19:
I created an example which you can clone from here: https://github.com/trashpirate/bullet_spring_constraint_demo/tree/master

Let me know if you have issues compiling it. Thanks!

erwincoumans, 2018-02-28:
Could you try attached fix? (unzip and copy over the old version)

[btGeneric6DofSpring2Constraint.cpp.zip](https://github.com/bulletphysics/bullet3/files/1769179/btGeneric6DofSpring2Constraint.cpp.zip)

trashpirate, 2018-03-08:
Sorry about the late response. I can confirm that the simulated frequencies now correspond with those of the btGeneric6DofSpringConstraint, thank you!

benelot, 2018-03-08:
Any reason why the better version is not merged into master? Or how come it
is packed in a zip instead?

erwincoumans, 2018-03-08:
Thanks for confirming!
@benelot 
It was already applied in a pull request here:
https://github.com/bulletphysics/bullet3/pull/1589/commits/86ca7dc2d13a38db031c762dee0ff091b13357ff

closed 2018-03-08 by erwincoumans