bulletphysics/bullet3 #1570: Bullet3 Windows DLL build does not export all data

madebr, 2018-02-21:
When trying to use `gcontactProcessedCallback`, defined [here](https://github.com/bulletphysics/bullet3/blob/62b86e6eab91ce82bcfd4f7461d41ecd4a52bc01/src/BulletCollision/NarrowPhaseCollision/btPersistentManifold.h#L37) and declared [here](https://github.com/bulletphysics/bullet3/blob/62b86e6eab91ce82bcfd4f7461d41ecd4a52bc01/src/BulletCollision/NarrowPhaseCollision/btPersistentManifold.cpp#L28), of a DLL build of bullet3, the linker fails to find that symbol.

The error is:
```sh
main.obj : error LNK2001: unresolved external symbol "bool (__cdecl* gContactProcessedCallback)(class btManifoldPoint &,void *,void *)" (?gContactProcessedCallback@@3P6A_NAEAVbtManifoldPoint@@PEAX1@ZEA) [C:\issues\conan_bullet_1\build\issue.vcxproj]
```

Linking to a static Bullet library is succesful.

## How to reproduce:

* I have created a minimum project:
  * It uses the [conan](https://www.conan.io/) build of bullet, defined [here](https://github.com/bincrafters/conan-bullet)
  * dynamic bullet library: https://github.com/madebr/issues/tree/master/conan_bullet_1
  * static bullet library: https://github.com/madebr/issues/tree/master/conan_bullet_2
  * Build it on Windows, using Visual Studio 2017

Failed build of a use of dynamic bullet library on appveyor: https://ci.appveyor.com/project/madebr/issues/build/1.0.24/job/8qbslvwoc1j2dbnh

Succeeded build of a use of static bullet library on appveyor: https://ci.appveyor.com/project/madebr/issues/build/1.0.24/job/hbgvx3uoxriqi3tp

Reference: bincrafters/community#130

erwincoumans, 2018-02-21:
I recommend using the static library, generally.

solvingj, 2018-02-21:
Should the “BUILD_SHARED_LIBS” cmake flag be removed if unsupported?

erwincoumans, 2018-07-27:
Some people use BUILD_SHARED_LIBS without problem on Linux and Mac OSX. We don't support BUILD_SHARED_LIBS on Windows. If someone wants to add support for BUILD_SHARED_LIBS, feel free to contribute.

closed 2018-07-27 by erwincoumans

pkovacs86, 2019-08-13:
I can confirm that the following cmake command could build the correct DLLs on Windows
if(NOT BULLET_DLL_READY)
# build bullet dlls
	foreach (CONF ${CMAKE_CONFIGURATION_TYPES})
		message (STATUS "Configuring bullet for ${CONF}, please wait.....")
		execute_process ( 
				COMMAND ${CMAKE_COMMAND}
				-DBUILD_SHARED_LIBS:BOOL=ON
				-DUSE_MSVC_RUNTIME_LIBRARY_DLL:BOOL=ON
				-DBUILD_BULLET2_DEMOS:BOOL=OFF
				-DBUILD_OPENGL3_DEMOS:BOOL=OFF
				-DBUILD_CPU_DEMOS:BOOL=OFF
				-DBUILD_UNIT_TESTS:BOOL=OFF
				-DBUILD_EXTRAS:BOOL=OFF
				-G ${CMAKE_GENERATOR} 
				.
				WORKING_DIRECTORY ${BULLET_HOME}
				RESULT_VARIABLE error OUTPUT_VARIABLE output ERROR_VARIABLE output
				)		
		reportError (output if error)
		message (STATUS "Building bullet for ${CONF}, please wait.....")
		execute_process ( 
			COMMAND ${CMAKE_COMMAND} --build . --config ${CONF} --target ALL_BUILD
			WORKING_DIRECTORY ${BULLET_HOME}
			RESULT_VARIABLE error OUTPUT_VARIABLE output ERROR_VARIABLE output)
		reportError (output if error)
	endforeach ()
endif ()

madebr, 2019-08-13:
@pkovacs86 
Please try building this executable and link it to your dll.
```
#include <BulletCollision/NarrowPhaseCollision/btPersistentManifold.h>
void main() {
    gContactProcessedCallback = nullptr;
}
```

The problem with dll's is that extern variables are not exported:
https://github.com/bulletphysics/bullet3/blob/cdd56e46411527772711da5357c856a90ad9ea67/src/BulletCollision/NarrowPhaseCollision/btPersistentManifold.h#L39-L42
and similar lines (probably all those using `extern`) should be exported.
In other words, they require a `__declspec(dllexport)` in Visual Studio and `__attribute__((visibility("default")))` in other compilers.

I guess it would be useful to prepend these and similar variables with BULLET3_API and define BULLET3_API using the some ifdef'ery.