bulletphysics/bullet3 #1576: kuka_diverse_object_gym_env: missing random_urdfs file

bwhite-envfluids, 2018-02-24:
The random_urdfs file that is used to load objects in kuka_diverse_object_gym_env.py seems to be missing from bullet3/examples/pybullet/gym/pybullet_data/. I found a technical note suggesting the file needed to be manually unzipped. However, even the random_urfs.zip file seems to be missing from the current implementation of pybullet. Thanks for any help you can give.

ericjang, 2018-03-04:
Thanks for bringing this to our attention. We'll check in the random_urdfs.zip shortly.

erwincoumans, 2018-03-04:
The files are all added directly, no need to unzip anymore:
https://github.com/bulletphysics/bullet3/tree/master/examples/pybullet/gym/pybullet_data/random_urdfs
The PyPi version already has all the files, check this out:

pip install pybullet
python -m pybullet_envs.baselines.enjoy_kuka_diverse_object_grasping

closed 2018-03-04 by erwincoumans