bulletphysics/bullet3 #1580: Tutorials are broken

elect86, 2018-02-27:
Every sample I try to run under `Tutorials` in the Example Browser throw this:

[image]

erwincoumans, 2018-02-28:
It works just fine here. Did you change your current working directory? The tutorial load a texture file checker_huge.gif, in a data directory in one of the parent directories of the current working directory.

closed 2018-02-28 by erwincoumans

elect86, 2018-02-28:
I just cloned the project and generated using `cmake` for VS 2017 Win64, am I not supposed to do like this?

erwincoumans, 2018-02-28:
It all depends from what directory you run cmake. If things don't work, first check if the provided script don't work either. If they work fine, compare what is different from your approach.