bulletphysics/bullet3 #1581: Cannot find URDF/SDF files for loadURDF/loadSDF

scchow, 2018-02-27:
Hi, this is my first time using Bullet so apologies if this is a simple fix.

I was following the pybullet quickstart guide, specificially the Hello pybullet world script:

```
import pybullet as p
import pybullet_data
physicsClient = p.connect(p.GUI)#or p.DIRECT for non-graphical version
p.setAdditionalSearchPath(pybullet_data.getDataPath()) #optionally
p.setGravity(0,0,-10)
planeId = p.loadURDF("plane.urdf")
cubeStartPos = [0,0,1]
cubeStartOrientation = p.getQuaternionFromEuler([0,0,0])
boxId = p.loadURDF("r2d2.urdf",cubeStartPos, cubeStartOrientation)
p.stepSimulation()
cubePos, cubeOrn = p.getBasePositionAndOrientation(boxId)
print(cubePos,cubeOrn)
p.disconnect()
```
This demo script worked.

I wanted to load a very simple URDF/SDF to verify that it works. I changed the loadURDF line from 
`boxId = p.loadURDF("r2d2.urdf",cubeStartPos, cubeStartOrientation)` to 
`boxId = p.loadURDF("~/<my_dir>/test_rob.urdf",cubeStartPos, cubeStartOrientation)`

and got the following error:
```
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,139]:

b3Printf: URDF file '~/<my_dir>/test_rob.urdf' not found

Traceback (most recent call last):
  File "hello.py", line 20, in <module>
    boxId = p.loadURDF("~/<my_dir>/test_rob.urdf",cubeStartPos, cubeStartOrientation)
pybullet.error: Cannot load URDF file.

```

Curious on why it can't find my urdf, I tested the script with other URDF and SDF files in the `data` directory. 

I noticed files like `r2d2.urdf` and `pr2_gripper.urdf` worked, but `pole.urdf` and `prismatic.urdf` don't. 

I was also hoping to try to load from SDF and encountered similar issues when I tried 
`boxId = p.loadSDF("two_cubes.sdf",cubeStartPos, cubeStartOrientation)`

I also tried placing my test_rob.urdf into the `data` directory, but it did not fix my issue.

Is there an issue with the search path that I am missing?

erwincoumans, 2018-02-28:
Things work just fine here, can you try unzipping and opening the two_cubes.py file?

Create a full reproduction similar to attached zip file: add the assets such as urdf, sdf, obj, mtl, png in the same directory as the python file or in a data folder as child of the directory where the python file, exactly like in attached example.

[two_cubes.zip](https://github.com/bulletphysics/bullet3/files/1767906/two_cubes.zip)

erwincoumans, 2018-02-28:
re-open if the issue persists.

closed 2018-02-28 by erwincoumans

khaledgg, 2022-10-20:
I am having the same issue, when trying to run the "two_cubes.py" file after downloading the "two_cubes.zip", it is still not loading resulting in the below error message

`b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,257]:

b3Printf: SDF file 'data/two_cubes.sdf' not found

Traceback (most recent call last):
  File "c:\Users\<path_to_folder>\two_cubes.py", line 4, in <module>
    cubes = p.loadSDF("data/two_cubes.sdf")
pybullet.error: Cannot load SDF file.` 

Any advice?