bulletphysics/bullet3 #1582: In btGeneric6DofConstraint the maximum motor force is used as an impulse despite the name

GaborPuhr, 2018-02-27:
In btGeneric6DofConstraint one can set the maximum motor force with the btRotationalLimitMotor::m_maxMotorForce and the btTranslationalLimitMotor::m_maxMotorForce.
Despite the name this value will be used as an Impulse (and not as a Force).
This is confusing and also the user should know the deltatime to set a correct value and that is not always easy to obtain.

Checking other constraints:
In case of btHingeConstraint it is correctly called setMaxMotorImpulse.
In case of btSliderConstraint it is called setMaxLinMotorForce and it is correctly used as force.

I suggest to change the implementation similar to btSliderConstraint by simply dividing with the FPS.

(The problem exist also in btGeneric6DofSpringConstraint and btGeneric6DofSpring2Constraint)

erwincoumans, 2018-02-28:
Thanks for bringing this up, it is best to fix this indeed (dividing by FPS when using a force).

GaborPuhr, 2018-03-01:
I will create a pull request shortly.

closed 2018-04-12 by erwincoumans