bulletphysics/bullet3 #1583: pybullet stall

BlGene, 2018-02-28:
I'm not sure this is a pybullet bug, but...
When training with pybullet training will occasionally stall, in this case after ~8M time steps. When I then terminate the process I get the following error message https://gist.github.com/BlGene/6614d30f39edff436beac19b4627cfea. This would suggest that calling `getCameraImage` fails.

Debugging this is probably a bit difficult, is there a way to get a more informative error message?
Timeout checks on function calls sound like a reasonable solution to this problem. These could be on the optimization side or the bullet side, where would be the best place to put this?

erwincoumans, 2018-02-28:
Terminating just shows where the program counter was, it doesn't mean that the code stalls there. Create an easy to reproduce case (and zip all files and attach here) and someone may look at it.

closed 2018-02-28 by erwincoumans

erwincoumans, 2018-02-28:
I recommend looking into KukaDiverseObjectEnv . You can try it out with a random policy using the following command:
python -m pybullet_envs.baselines.enjoy_kuka_diverse_object_grasping
The KukaCamGymEnv hasn't been used/tried out yet, so it could have issues, but we never have seen issues with getCameraImage, stall is more likely in the training code.