bulletphysics/bullet3 #1586: Reprojection of depth image giving incorrect values

puneetp, 2018-03-03:
I am trying to find the 3D point based on the depth scan. I need this to redirect kuka arm to a certain location (based on point identified on depth image)

I am following the standard projection formula where image pixel being reprojected is at (u,v) and focal length(f) is assumed to be the first element of Projection matrix at (0,0)

cx=320 // principle pixel x value
cy=240 // principle pixel y value
X=(z/f)*(u-cx)   // z is the depth value after converting to real world dimensions using [1]  
Y=(z/f)*(v-cy)
Z=z 
3D_PointinWorldFrame = inv(ViewMatrix)* [X;Y;Z;1] //

Questions
My 3D point is not matching where the object is placed and is off by significant amount. 
a) Is there any offset I am missing or are there any units incorrectly being used which may be causing this serious reprojection error?
b) Do you see any issues with the steps followed above Or are there any camera model bugs in the code?
(I chose the near value to be small 0.001  to avoid the issue #1197 )

I am working on KukaGymEnv.py (using the _render function) and using it for RL work

Thanks

--------------------------------------------------------------

Values in case you want to test it out
Source Point Object in real world is at  (x,y,z): (0.634, 0.0, -0.19)
Projection Matrix =  (1.2990381717681885, 0.0, 0.0, 0.0, 0.0, 1.732050895690918, 0.0, 0.0, 0.0, 0.0, -1.0000020265579224, -1.0, 0.0, 0.0, -0.00020000019867438823, 0.0)
View Matrix  (-1.0, 8.742278367890322e-08, -0.0, 0.0, -8.742278367890322e-08, -1.0, -0.0, 0.0, -0.0, 0.0, 1.0, 0.0, 0.75, -0.10000006854534149, -0.5699999928474426, 1.0)
(FarValue=100,NearValue=0.0001)
On Depth Image the xy and depth value is (378,296,0.679)
The 3D_PointinWorldFrame after doing the reprojecting calculation is at (x/w,y/w,z/w): (-1.430,-1.3808,0.0320)

As you can see it is very different from source point values

[1] http://web.archive.org/web/20130416194336/http://olivers.posterous.com/linear-depth-in-glsl-for-real

erwincoumans, 2018-03-03:
The depth issue was fixed previously, I just closed that issue (see https://github.com/bulletphysics/bullet3/issues/1197)

You need to use the full inverse computations to unproject from pixel location/depth to 3D position, your formula doesn't seem complete. Good luck fixing your unprojection, and please consider sharing it when you are done.

erwincoumans, 2018-03-03:
I'll keep the issue open for a little while, please spend some time on it and get back if you need specific help.

erwincoumans, 2018-03-03:
The previous issue was resolved a while ago, the depth issue was fixed previously, I just closed that issue (see Issue #1197)

That issue has an example to compute the depth from the OpenGL depth image, it involves both near and far plane distance, your computation doesn't.

In the example, the camera eye is looking from [0,0,1] downward at a plane at location [0,0,0] pointing up and a little cube of 5cm with its center at position 0.025:

~~~~~~
import matplotlib.pyplot as plt
import numpy as np
import pybullet as p

p.connect(p.GUI)
p.loadURDF('plane.urdf')
p.loadURDF('cube_small.urdf', basePosition=[0.0, 0.0, 0.025])

width = 128
height = 128

fov = 60
aspect = width / height
near = 0.02
far = 100

view_matrix = p.computeViewMatrix([0, 0, 1], [0, 0, 0], [1, 0, 0])
projection_matrix = p.computeProjectionMatrixFOV(fov, aspect, near, far)

# Get depth values using the OpenGL renderer
images = p.getCameraImage(width, height, view_matrix, projection_matrix, renderer=p.ER_BULLET_HARDWARE_OPENGL)
depth_buffer_opengl = np.reshape(images[3], [width, height])
depth_opengl = far * near / (far - (far - near) * depth_buffer_opengl)

# Get depth values using Tiny renderer
images = p.getCameraImage(width, height, view_matrix, projection_matrix, renderer=p.ER_TINY_RENDERER)
depth_buffer_tiny = np.reshape(images[3], [width, height])
depth_tiny = far * near / (far - (far - near) * depth_buffer_tiny)

p.disconnect()

# Plot both images - should show depth values of 0.45 over the cube and 0.5 over the plane
plt.subplot(1, 2, 1)
plt.imshow(depth_opengl, cmap='gray', vmin=0, vmax=1)
plt.title('OpenGL Renderer')

plt.subplot(1, 2, 2)
plt.imshow(depth_tiny, cmap='gray', vmin=0, vmax=1)
plt.title('Tiny Renderer')

plt.show()
~~~~~~

closed 2018-03-03 by erwincoumans

erwincoumans, 2018-03-11:
I had some time to create an example that renders a camera image and then projects the pixels back into world space. It uses some helper function in C++, so you need to still do some extra work. See the [example code here](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/pointCloudFromCameraImage.py). The implementation of getDebugVisualizerCamera in C++ is in [OpenGLGuiHelper::getCameraInfo](https://github.com/bulletphysics/bullet3/blob/master/examples/ExampleBrowser/OpenGLGuiHelper.cpp#L1045)

[image]

[image]