bulletphysics/bullet3 #1600: PosixSharedMemory.cpp doesn't use Posix methods.

lshamis, 2018-03-14:
shmget, shmat, shmctl, and shmdt are "System V" methods for ipc.
Posix uses shm_open (+ftruncate +mmap).

This is an issue for Android devices.

erwincoumans, 2018-03-14:
Ah, that's right. Note that by default (DIRECT and GUI mode), PyBullet and the Bullet C API use InProcessMemory. So you should be able to use the software on Android, just without PosixSharedMemory.

lshamis, 2018-03-14:
I was able to remove PosixSharedMemory from the linker dependencies and needed to comment out: https://github.com/bulletphysics/bullet3/blob/282458684fa2e56f33b22e3ccb011d2c1c3d4f25/examples/RobotSimulator/b3RobotSimulatorClientAPI_NoGUI.cpp#L76

Feature request: Guard shared memory in the same way udp is guarded by BT_ENABLE_ENET and tcp is guarded by BT_ENABLE_CLSOCKET.

erwincoumans, 2018-03-15:
We can add a BT_DISABLE_SHARED_MEMORY option around the case
(I don't have time to update all the build files here internally, so a BT_ENABLE_SHARED_MEMORY is not going to happen). If you feel like, please add it and create a pull request.

erwincoumans, 2018-12-23:
We don't have resources to work on this. If a volunteer shows up, please re-open and consider a pull request.

closed 2018-12-23 by erwincoumans