bulletphysics/bullet3 #1608: Multiple additional search paths for PyBullet

Kukanani, 2018-03-19:
Is there a way to specify multiple additional search paths in PyBullet? From my reading of `setAdditionalSearchPath`, the last-provided path overrides all previous paths. I tried concatenating paths with `;`, `:`, and `,`, as well as providing a list or tuple of paths, but none of these resulted in the behavior I was hoping for, which was searching multiple paths for assets.

erwincoumans, 2018-03-19:
At the moment it is not possible. For an 'addAdditionalSearchPath' we would need either a fixed number of total search path, or things get more complicated with memory management.
How many paths do you need? Can't you move/copy the data in a similar root folder?

Kukanani, 2018-03-19:
Thanks for the quick response.

When I hear "search path" my mind goes to system $PATH variables, which can contain any number of paths. In my specific case, I'm trying to search in both the `pybullet_data` folder, and a folder specific to my project. I can think of other scenarios where you'd want to allow multiple search paths. For example, if your data files are stored in separate places (say you're using URDFs from upstream GitHub repos) and you don't want your data to be desynced with the other repositories.

Your comment about memory management makes more sense to me now that I see that pybullet is written in C. I don't know what it would take to properly handle a Python list of strings as input but I can guess that it would not be straightforward.

erwincoumans, 2018-12-23:
With the new fileIOPlugin you can set additional search path, and also load from zip files etc.
See also fileIOPlugin.py example.

closed 2018-12-23 by erwincoumans