bulletphysics/bullet3 #1618: Undefined gravity behaviour

1nir1, 2018-03-28:
Hello,
We are trying to create a ball and move it around the stadium.sdf environment using pybullet, but we are having an issue - we aren't sure it is related to pybullet directly but we will appreciate if you can help.
When using the module:
```
<mujoco model="WB">
<compiler angle="degree" coordinate="local" inertiafromgeom="true"/>
<option integrator="RK4" timestep="0.01"/>
<default>
<joint limited="true"/>
<geom friction="1 0.5 0.5"/>
</default>
<worldbody>
<body name="main_torso" pos="0 0 1.25">
<geom name="my_g" fromto="0 0 0.4" type="sphere" size="0.25"></geom>
<joint name="j_root" axis="0 0 1" pos="0 0 0" range="-360 360" type="hinge"></joint>
</body>
</worldbody>
<actuator>
<motor ctrllimited="false" crtlrange="-1.0 1.0" joint="j_root" gear="110"/>
</actuator>
</mujoco>
```
The result is:
[image]

And the ball stays stuck up there - when we try to move it, it just moves in it's place. Why doesn't it just fall down and move around the environment? Are we missing something?

Thank you.

erwincoumans, 2018-03-28:
Where is the PyBullet script you used? Did you set the gravity using pybullet.setGravity(x,y,z)?
Also see the [PyBullet Quickstart Guide](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.2ye70wns7io3).

1nir1, 2018-03-28:
We created a similar PyBullet environment like in the examples. We set the following gravity: (0, 0, -9.8). Attached to this message another example we ran with the same PyBullet script. In this case the ball fall down, but we can't move him around. It seems like it is somehow related to the first joint we add to the body? Can it be? 
We have looked at the PyBullet guide, but we couldn't find any solution. Regarding, the URDF, we prefer to use the most basic XML format we can. We only used the headline of MJCF because the example we initially built our project on was Ant.

The example:

> <mujoco model="WB">
    <compiler angle="degree" coordinate="local" inertiafromgeom="true"/>
    <option integrator="RK4" timestep="0.01"/>
    <default>
        <joint limited="true"/>
        <geom friction="1 0.5 0.5" />
    </default>
    <worldbody>
        <body name="main_torso" pos="0 0 1.25" >
            <geom name="my_g" fromto="0 0 0.4" type="sphere" size="0.25" >
            </geom>
            <body name="torso" pos="0 0 0" >
                <joint name="j_root" axis="1 1 1" pos="0 0 0" range="-360 360" type="hinge" >
                </joint>
            </body>
        </body>
    </worldbody>
    <actuator>
        <motor ctrllimited="false" crtlrange="-1.0 1.0" joint="j_root" gear="110" />
    </actuator>
</mujoco>

erwincoumans, 2018-03-28:
Try unlocking all motors after loading the robot:
~~~~~~~~~~
robot = p.loadMJCF("file")
numJoints = p.getNumJoints(robot)
for j in range (numJoints):
  p.setJointMotorControl2(robot,j, p.VELOCITY_CONTROL,force=0)
~~~~~~~~~~

1nir1, 2018-03-29:
We have read the quickstart guide and played with the examples - we posted this issue only after not finding a solution there. In any case - this is the code that we start our simulation run with:

>     def startRun(self, filePath = None, sdfPath = None, gravity = (0,0,-9.8), visual = True):
        if filePath is None:
            print ("Filepath is required")
            return
        if sdfPath is None:
            sdfPath = getDefaultSDF()

        self.finishBuild = True
        self.WB.prettyPrint(filePath)

        cid = p.connect(p.SHARED_MEMORY)
        if (cid<0):
            if visual is True:
                cid = p.connect(p.GUI)
            else:
                cid = p.connect(p.DIRECT)

        p.setGravity(gravity[0], gravity[1], gravity[2])
        p.setPhysicsEngineParameter(fixedTimeStep=1.0/60., numSolverIterations=5, numSubSteps=2)

        p.loadSDF(sdfPath)
        objs = p.loadMJCF(filePath + ".xml" , flags = p.URDF_USE_SELF_COLLISION_EXCLUDE_ALL_PARENTS)

        self.robot  = objs[0]
        ordered_joints = []
        ordered_joint_indices = []

        parser = argparse.ArgumentParser()
        parser.add_argument('--profile')
        self.jdict = {}
        for j in range( p.getNumJoints(self.robot) ):
            info = p.getJointInfo(self.robot, j)
            ordered_joint_indices.append(j)
            
            if info[2] != p.JOINT_REVOLUTE: continue
            jname = info[1].decode("ascii")
            self.jdict[jname] = j
            lower, upper = (info[8], info[9])
            ordered_joints.append( (j, lower, upper) )
            p.setJointMotorControl2(self.robot, j, controlMode=p.VELOCITY_CONTROL, force=0)

        self.frame = 0
        self.startTime = time.time()

        self.startPosition, (qx, qy, qz, qw) = p.getBasePositionAndOrientation(self.robot)
        print (self.startPosition)

As you can see we are unlocking all motors first. Any idea why still the body acts so rigid?

erwincoumans, 2018-03-29:
Please provide a small reproduction case with only pybullet as dependency and someone may look at it. Again, compare why other examples work fine and yours doesn't.

1nir1, 2018-04-01:
Attached a small example that demonstrates the issue we talked about - with both XML examples we provided before. Only other dependency are time - which we used for intervals between simulation steps and argparse that you have in your examples - we didn't remove it.
[Example.zip](https://github.com/bulletphysics/bullet3/files/1866672/Example.zip)

erwincoumans, 2018-04-01:
The attached file is not a valid XML file (MuJoco PRO 1.50 doesn't load it).
Aside from that, the sphere in the XML file is attached to the world and can only rotate, since you added a 'hinge' joint.
Attached is an example that lets you control a freely moving sphere.
[controlSphere.zip](https://github.com/bulletphysics/bullet3/files/1866758/controlSphere.zip)