bulletphysics/bullet3 #1620: Random motion in nao bot when played on mujoco

JoyChopra1298, 2018-04-02:
I ran the nao.urdf using mujoco `simulate` binary. There are many random motions in the simulation. I think this is because of the contact forces among a few meshes that are too close like torso and shoulder. When I removed all the contact forces, the random motion vanished.
 I have attached a gif that shows the problem
[image]

erwincoumans, 2018-04-02:
We support open source PyBullet here, it should simulate the nao.urdf robot with contact forces just fine. If you use mujoco (why?) please visit their forums to report such issues.

closed 2018-04-02 by erwincoumans

erwincoumans, 2018-04-03:
Nao came somewhere from github, search for it.
Attached is a small script that simulated nao in pybullet (use pip install pybullet first)
Note that we can enable self-collision, but we need to exclude child-parent collision for directly connected links, since they overlap.
[nao.zip](https://github.com/bulletphysics/bullet3/files/1872288/nao.zip)

OlalekanIsola, 2020-03-11:
Hi, i am new to mujoco and robotics.Could you please walk me  through setting it up.I have installed mujoco on Ubuntu 16.04

erwincoumans, 2020-03-11:
@OlalekanIsola pip3 install --upgrade pybullet --user

OlalekanIsola, 2020-03-12:
Thank you so much for your response, i have done the pip install. Could you please give detailed step by step procedure for running NAO.Thanks

OlalekanIsola, 2020-03-12:
i ran nao.py but i got the error below

ola@ola-HP-Pavilion-15-Notebook-PC:~/Downloads$ python nao.py
pybullet build time: Feb 22 2020 01:51:06
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.3 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=Intel Open Source Technology Center
GL_RENDERER=Mesa DRI Intel(R) Haswell Mobile 
GL_VERSION=3.3 (Core Profile) Mesa 18.0.5
GL_SHADING_LANGUAGE_VERSION=3.30
pthread_getconcurrency()=0
Version = 3.3 (Core Profile) Mesa 18.0.5
Vendor = Intel Open Source Technology Center
Renderer = Mesa DRI Intel(R) Haswell Mobile 
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
ven = Intel Open Source Technology Center
Workaround for some crash in the Intel OpenGL driver on Linux/Ubuntu
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,152]:

b3Printf: URDF file 'plane.urdf' not found

Traceback (most recent call last):
  File "nao.py", line 5, in <module>
    p.loadURDF("plane.urdf")
pybullet.error: Cannot load URDF file.
numActiveThreads = 0
stopping threads
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed
finished
numActiveThreads = 0
btShutDownExampleBrowser stopping threads
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed

vishwascmouly, 2022-04-15:
> Thank you so much for your response, i have done the pip install. Could you please give detailed step by step procedure for running NAO.Thanks

@OlalekanIsola  @erwincoumans can u please provide the nao robot pybullet files. I am not able to find them on the internet, the files I found are showing errors like "cannot load urdf file".

duhaoyuan1111, 2022-10-17:
> Thanks for the quick reply. Can you tell how did you make the nao.urdf or give contact of the person who made it? It will help me a lot.

Hello, I got the exactly same issue as you had, we wanna stay with Mujoco, but modify this xml file so that it won't be colliding (shaking) all the time. Have you solved this issue? Thanks a lot!

nico-bohlinger, 2023-05-28:
> Hello, I got the exactly same issue as you had, we wanna stay with Mujoco, but modify this xml file so that it won't be colliding (shaking) all the time. Have you solved this issue? Thanks a lot!

Hey, did you find a solution or have you modified the xml file?