bulletphysics/bullet3 #1623: Wrong depth reported between sphere and trimesh collision

AndreaCatania, 2018-04-04:
## Overview
This issue was found on godot engine, while I was checking what was happening here: https://github.com/godotengine/godot/issues/17960

I've noticed that when my kinematic body was using a sphere shape, sometimes the engine was returning me a positive depth result even if the shape is penetrated.

## Details
To check the collision between kinematic body with sphere shape and trimesh shape I'm using this algorithm: https://github.com/godotengine/godot/blob/master/modules/bullet/space_bullet.cpp#L1100-L1136

and here the manifold result class to collect the data: https://github.com/godotengine/godot/blob/master/modules/bullet/godot_result_callbacks.cpp#L258-L268

## Problem
As you can see from the addContactPoint function from the ManifoldResult callback (link above) I've this like of code:

`if (m_penetration_distance > depth) { // Has penetration?` 

That check if there was penetration, since the depth is negative if penetration happens m_penetration_distance value is more than depth value (initially m_penetration_distance is 0).

Even if the shape is penetrated for some reason the depth is positive. This start happening after a bit in I'm moving the kinematic on certain areas of this trimesh shape, and more important thing is that if I change the kinematic shape from sphere to capsule this problem never happens.

## Run the project:
Compile godot engine from master branch and open this project: 
https://github.com/godotengine/godot/files/1874984/godot-desert-physics_collision_error.zip

Move the camera using W-A-S-D and arrows of keyboards, move the kinematic body using T-F-G-H

## Images
In this images the kinematic body is stuck
The blue line is the normal of contact point

Screenshot from the above
[image]

Screenshot from the below to assert that there's penetration
[image]

Screenshot from the above more close
[image]

erwincoumans, 2018-04-04:
Hi Andrea,

Thanks a lot for reporting an issue. I'm happy to look at it within Bullet: can you please create a reproduction case using plain Bullet, in a way that doesn't require user interaction etc?

Bullet has a way to serialize (part of) the state with just a few lines of code:
Ideally you just save the state of the sphere and the triangle mesh. Since there are many ways to create a sphere and a triangle mesh in Bullet, and it could be due to certain settings, it is important to create a simple reproduction case.

Here is some info how to serialize to a binary .bullet file:
http://bulletphysics.org/mediawiki-1.5.8/index.php/Bullet_binary_serialization

Please let me know if you run into issues creating a reproduction case.
Thanks!
Erwin

AndreaCatania, 2018-04-04:
Do you need only the serialized .bullet file? or do I have to implement it in a "plain" project with only bullet?

Also do you need only kinematic body serialization and the "terrain" or the entire world is better?

erwincoumans, 2018-04-04:
Some main.cpp that loads the .bullet file, does a query (btCollisionWorld::contactPairTest?) :that shows "a positive depth result even if the shape is penetrated.".

We basically need to isolate the problem. Does it also happen when using btCollisionWorld::contactPairTest?

AndreaCatania, 2018-04-05:
I've:
- extracted the .bullet file for both shape sphere and trimesh.
- position of sphere where the collision error happens
- created a new project that use only bullet.

But I can't understand how to load/create the shapes from .bullet file, can you please explain what I've to do to load the shapes?

AndreaCatania, 2018-04-05:
Here the project https://github.com/AndreaCatania/TestBullet/blob/master/main.cpp#L93 sorry if it's too dirty I did it more fast I could

Also the .bullet files 
https://github.com/AndreaCatania/TestBullet/blob/master/sphere.bullet
https://github.com/AndreaCatania/TestBullet/blob/master/trimesh.bullet

erwincoumans, 2018-04-05:
Thanks. here is some snippet that can load a shape.
~~~~~~~~~~~~~
btBulletWorldImporter* importer = new btBulletWorldImporter(m_dynamicsWorld);
	importer->loadFile("sphere.bullet");
	btAssert(importer->getNumCollisionShapes() == 1);
	if (importer->getNumCollisionShapes() == 1)
	{
		sphere = importer->getCollisionShapeByIndex(0);
	}
~~~~~~~~~~~~~
It is better to start modifying the BasicExample.cpp, so we can also visualize it in BulletExampleBrowser, see attached.
[BasicExample.zip](https://github.com/bulletphysics/bullet3/files/1881184/BasicExample.zip)

The trimesh.bullet doesn't contain a triangle mesh mesh.
Could you serialize the entire world instead of separate shapes
(use dynamicsWorld->serialize).

Alternatively, programatically re-create the problem (instead of using .bullet files).

AndreaCatania, 2018-04-09:
In this zip file you can find the two shapes (sphere and trimesh terrain) and the entire world
[world.zip](https://github.com/bulletphysics/bullet3/files/1889386/world.zip)

During separation process (of kinematic body) I'm using another shape that is a bit bigger, it's not part of world, so you will not find them in the world.bullet file

erwincoumans, 2018-04-12:
Do you manage to recreate the issue with this world file?

AndreaCatania, 2018-04-12:
Not yet, I got some problem during compilation of bullet with examples.. It was a problem on linking libraries, now I don't remember exactly. I'll check it soon..

AndreaCatania, 2018-04-12:
Since I'm able to debug Bullet code from Godot code and it's really easy, If you are ok to check it with-in godot I can give you a step by step list in order to debug it.

I've already the project setup in godot with kinematic body that start from the position where the body returns the positive depth.

erwincoumans, 2018-04-12:
I cannot debug godot code, I really need an isolate repro case without
customer code.

Please try to compile unmodified Bullet, then modify the BasicDemo.cpp step
by step.

AndreaCatania, 2018-05-02:
Hi Erwin, I'm really sorry for this delay but I was a bit busy with a moving house these two weeks... Now I'm managing to create the reproducible project by editing the BasicExample.cpp.

Can you please tell me the step that I've to do to include `btBulletWorldImporter.h`and link it fron the BasicExample demo correctly?

erwincoumans, 2018-05-02:
The BasicExample.zip includes the BasicExample.cpp that includes the btBulletWorldImporter.
You can use premake or cmake to generate build files for any platform (Linux, Mac OSX, Windows) and just compile the ExampleBrowser.  The ExampleBrowser starts with BasicExample by default. Just make sure to put the sphere.bullet and trimesh.bullet files in the current working directory (or use an absolute path)

AndreaCatania, 2018-05-03:
It's more or less the same thing that I did, but I'm getting the exact same compiling error:

```
$ make
[ 10%] Built target BulletCollision
[ 15%] Built target BulletDynamics
[ 17%] Built target LinearMath
[ 17%] Built target App_HelloWorld
[ 17%] Built target Bullet3Common
[ 19%] Built target OpenGLWindow
[ 19%] Linking CXX executable AppBasicExampleGui
CMakeFiles/AppBasicExampleGui.dir/BasicExample.o: In function `BasicExample::initPhysics()':
BasicExample.cpp:(.text+0x2b2): undefined reference to `btBulletWorldImporter::btBulletWorldImporter(btDynamicsWorld*)'
BasicExample.cpp:(.text+0x2c1): undefined reference to `btBulletWorldImporter::loadFile(char const*, char const*)'
BasicExample.cpp:(.text+0x2cc): undefined reference to `btWorldImporter::getNumCollisionShapes() const'
BasicExample.cpp:(.text+0x2e4): undefined reference to `btBulletWorldImporter::loadFile(char const*, char const*)'
BasicExample.cpp:(.text+0x2f1): undefined reference to `btWorldImporter::getNumCollisionShapes() const'
BasicExample.cpp:(.text+0x4a1): undefined reference to `btWorldImporter::getCollisionShapeByIndex(int)'
BasicExample.cpp:(.text+0x6b6): undefined reference to `btWorldImporter::getCollisionShapeByIndex(int)'
collect2: error: ld returned 1 exit status
make[2]: *** [examples/BasicDemo/CMakeFiles/AppBasicExampleGui.dir/build.make:167: examples/BasicDemo/AppBasicExampleGui] Error 1
make[1]: *** [CMakeFiles/Makefile2:308: examples/BasicDemo/CMakeFiles/AppBasicExampleGui.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

```

AndreaCatania, 2018-05-03:
I've searched inside CMake properties but all seems correct to me (at the end I'm using the default settings) Also I think that if you are able to compile it can't be a problem of Make file that doesn't include the cpp files....

Do you have any idea?

erwincoumans, 2018-05-03:
No, you should not compile ALL target, only  1 target:
~~~~~~~
make App_ExampleBrowser
~~~~~~~

The BasicExample.cpp is used in several targets, and most of them don't include the required dependencies (for example, AppBasicExampleGui doesn't link against BulletWorldImporter).
The App_ExampleBrowser target also includes BasicExample.cpp, and it has the required dependencies.

AndreaCatania, 2018-05-04:
Thanks you for your help! In this archive you can find the scene and the trimesh.bullet. the scene has trimesh shape and sphere at the position where the depth returned is positive.

The query is done inside the function `renderScene`

I want to point out this, the camera is pointing from  above to the trimesh and the trimesh normals are "reversed" you can notice that the trimesh is "invisible", due to the culling. I don't know if this could be the problem (in godot it's show correctly). However even if the normals are inthe opposite direction the depth should be negative

[BasicExample.zip](https://github.com/bulletphysics/bullet3/files/1974749/BasicExample.zip)

AndreaCatania, 2018-05-14:
Hi Erwin, did you got the time to look into it?

AndreaCatania, 2018-05-14:
It's fine, thanks you!!

erwincoumans, 2018-05-22:
Should be fixed by this commit: https://github.com/bulletphysics/bullet3/pull/1695

closed 2018-05-22 by erwincoumans