bulletphysics/bullet3 #1626: Disable joints for inverse kinematic solution

lukashermann, 2018-04-05:
Hi everyone,
I'm trying to do cartesian control in pybullet with a model of the PR2 robot. Currently I would like to use only the right arm of the robot, but when I calculate the pybullet inverse kinematics for the end-effector, the robot (naturally) uses also other joints (e.g. torso lift up) to achieve the desired pose. Is there an elegant way to tell pybullet to only use certain joints for the inverse kinematics, so to say regard the rest as fixed?
Thanks in advance!

erwincoumans, 2018-04-12:
Selecting the root of IK is currently not exposed (but it could be implemented, the underlying IK is very flexible/powerful). 

Alternatively, you could try to create a similar model in a separate simulation (note that PyBullet can connect to multiple 'DIRECT" simulation instances that you can use for this), and have the joints fixed.

Finally, you could try using a very high damping in the IK, for the torso joints.
This damping works for IK2_DLS, IK2_VEL_DLS_WITH_ORIENTATION, IK2_VEL_DLS

lukashermann, 2018-04-16:
Using a high damping for the joints except the arm joints results in a very slow movement of the arm itself, although I'm not sure why. The desired pose is never reached. However, I found another workaround: computing the IK for all joints, but then setting the jointPoses of the rigid joints to zero seems to work quite well.

closed 2018-04-16 by lukashermann