bulletphysics/bullet3 #1627: Segmentation faults when importing Baxter URDF

maxspero, 2018-04-05:
I'm having trouble importing a URDF for the Rethink Baxter robot into Bullet.

Here is the gist of the code I am trying to run:
```
import pybullet as p
p.connect(p.DIRECT)
robot = p.loadURDF("baxter.urdf")
```

And it results in a segmentation fault:
```
$ python urdf.py
pybullet build time: Mar  1 2018 17:05:08
b3Warning[/Users/maxspero/Documents/svl/bullet3/examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,114]:
No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frameb3Warning[/Users/maxspero/Documents/svl/bullet3/examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,114]:
baseb3Warning[/Users/maxspero/Documents/svl/bullet3/examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,774]:
data/baxter_common/baxter_description/urdf/: cannot extract mesh from 'data/baxter_common/baxter_description/meshes/torso/base_link_collision.DAE'
Segmentation fault: 11
```

It seems these DAE files don't play well with bullet. Removing the two collision meshes that cause segfaults allows us to render the arms. 

[image]

However, it comes with a bunch of warnings about the remaining DAE meshes:

```
b3Printf: ./data/baxter_common/baxter_description/urdf/: cannot extract anything useful from mesh './data/baxter_common/baxter_description/meshes/*/*.DAE'

b3Printf: b3Warning[/Users/maxspero/Documents/svl/bullet3/examples/SharedMemory/plugins/tinyRendererPlugin/TinyRendererVisualShapeConverter.cpp,440]:

b3Printf: issue extracting mesh from COLLADA/STL file ./data/baxter_common/baxter_description/meshes/*/*.DAE
```

See the code used and URDF/meshes [here](https://drive.google.com/file/d/1X7ZWvkiPgUnOx4j8pYw6jSlEQEqze8s7/view?usp=sharing).

Any ideas on how to import this URDF more smoothly? Thank you!

erwincoumans, 2018-04-05:
Could you try converting the COLLADA files into .obj or stl? COLLADA is a very complex file format, and Bullet only supports a tiny subset of it.

maxspero, 2018-04-07:
Thanks, that worked! It seems like it was just incompatibility with the COLLADA files.

closed 2018-04-07 by maxspero

erwincoumans, 2018-04-07:
Great. Do you mind sharing the updated Baxter URDF/obj files? 
I could add them to 
 https://github.com/erwincoumans/pybullet_robots
It has the Sawyer, Atlas, Dobot and some other robots.

maxspero, 2018-04-07:
[Here you go!](https://drive.google.com/open?id=16vXFfig3XkonBwCHeACSeHNO9iVWxMO_)