bulletphysics/bullet3 #1631: Clarify link labels with ER_SEGMENTATION_MASK_OBJECT_AND_LINKINDEX in documentation

cpaxton, 2018-04-11:
It looks like objects with only one link (free-floating rigid bodies) do not follow the rule in the documentation:

>  If ER_SEGMENTATION_MASK_OBJECT_AND_LINKINDEX is used, the segmentationMaskBuffer combines the object unique id and link index as follows:
value = objectUniqueId + (linkIndex+1)<<24. See example.

It would be great to update the documentation if this is true.

erwincoumans, 2018-04-11:
A free-floating body with only a base with no joints has no links. The base link index is -1. 
What is the issue?

cpaxton, 2018-04-11:
@erwincoumans  I understand this. I'm not personally having an issue, I thought that was unclear in the [documentation](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#). Also where does it say the base link index for all objects is -1?

erwincoumans, 2018-04-11:
Page 17 of the quickstart guide:
"Since the base is not a regular 'link', we use the convention of -1 as its link index."
In other words, for a free floating body, the segmentation mask is just its object unique id.
I'll add this line to the doc, does that help/explain things?

erwincoumans, 2018-04-11:
Thanks for the feedback, I added some comment, so let's close the issue.

closed 2018-04-11 by erwincoumans