bulletphysics/bullet3 #1633: Subtle Memory Leak in PyBullet

MichaelLutter, 2018-04-12:
When running the attached example, memory usage will steadily increase. The increase in memory consumption is most likely caused by pybullet.setJointMotorControlArray. For normal simulations this might not be an problem, but for Reinforcement Learning tasks, where the task is repeated thousands of times, this memory leak leads to freezing the computer. This was test on Ubuntu 16.04 and MacOs with pybullet 1.9.4.

Btw. I noticed that connecting via GUI can be 20 - 50% faster than connecting via DIRECT. This seems counterintuitive for me as GUI requires the additional rendering of the GUI. Is there an explanation why DIRECT is slower? 

[MemoryLeakPybullet.zip](https://github.com/bulletphysics/bullet3/files/1904353/MemoryLeakPybullet.zip)

erwincoumans, 2018-04-12:
Reinforcement Learning tasks are the primary use of PyBullet, so we should fix any leaks.
(although recently we start exploring [ARS](https://arxiv.org/abs/1803.07055), it seems to give faster results, compared to RL/PPO, DDPG, TRPO etc)

Thanks for the repro case!

erwincoumans, 2018-04-12:
Should be fixed now, thanks for the report!

closed 2018-04-12 by erwincoumans