bulletphysics/bullet3 #1636: Collision detection/ bounding box specification error

mjm522, 2018-04-15:
Hi, 

I am making a setup for a robot peg-insertion problem. A wide hole is constructed for both the visual shape and the collision shape. (specified by an STL file). But when the robot tries to penetrate the hole, the collision checking prevents it from insertion (can be seen here - https://youtu.be/ku-9KGcH_PQ). From the simulation, I understand that the collision shape taken is bounding the entire box instead of taking the STL file. How to prevent this?

my urdf description:

```
  <link name="sphereLink">
  <contact>
      <lateral_friction value="1.5"/>
  </contact>
    <inertial>
      <origin rpy="0 0 0" xyz="0 0 0"/>
       <mass value="0.5"/>
       <inertia ixx="0.1" ixy="0" ixz="0" iyy="0.1" iyz="0" izz="0.1"/>
    </inertial>
    <visual>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
	              <mesh filename="peg_hole2.STL" scale="0.01 0.01 0.01"/>
      </geometry>
       <material name="grey">
        <color rgba="0.5 0.5 0.5 1"/>
      </material>
    </visual>
    <collision>
      <origin rpy="0 0 0" xyz="0 0 0"/>
      <geometry>
	 	     <mesh filename="peg_hole2.STL" scale="0.01 0.01 0.01"/>
      </geometry>
    </collision>
  </link>
```

Can't this sort of setup cannot be made in bullet? If so how to specify the collision information as required? Could anyone please help? Thanks alot for your time.

Best,
Michael

erwincoumans, 2018-04-16:
There are a few ways: if the concave part is not a rigid body, but static, use this flag to collide against the triangle mesh:
~~~~~~~~~~~
<link concave="yes" name="baseLink">
~~~~~~~~~~~
Alternative, if you need the concave part moving too, create a convex decomposition, and insert each part as a separate object in the obj file. You can use V-HACD to create a convex decomposition.

I am currently working on adding signed distance fields in Bullet, it will be exposed in URDF too.
I'll report back when this is available.

mjm522, 2018-04-16:
thanks a lot for the reply. I am trying the methods you suggested.

mjm522, 2018-05-10:
A quick update, I found this python library [trimesh](https://github.com/mikedh/trimesh) to be extremely useful. If the [v-hacd](https://github.com/kmammou/v-hacd) library is already installed, then update the environment variable PATH to the testVHACD executable. Once that is done, 

```
import trimesh
from trimesh.io.urdf import export_urdf
mesh = trimesh.load(<path to STL/ OBJ/PLY file>)
export_urdf(mesh, <folder path to which urdf has to be saved>)
```
This will do a hierarchial decomposition of the concave object and generate a urdf file from the same. This urdf file can be easily used in bullet. 

I am on Ubuntu and hence I compiled the v-hacd. My export path is as follows:

` export PATH=$PATH:' <path to v-hacd>/build/linux2/test`

Note: The v-hacd have the binary for OSx and Windows, for Linux it has to be compiled using the the run.py in the scripts folder and then running make in the build directory. The build directory will be automatically created upon successful running of run.py

closed 2021-04-26 by erwincoumans