bulletphysics/bullet3 #1638: SHARED_MEMORY_SERVER support

YunfeiBai, 2018-04-16:
Having an equivalent version of the function below for headless mode, to start an in-process physics server, which PyBullet can connect to
https://github.com/bulletphysics/bullet3/blob/a506879d6cfd44f7df041d4fd05b3641fd1ed713/examples/SharedMemory/SharedMemoryInProcessPhysicsC_API.h#L14

mbusy, 2020-02-17:
Hi, any updates on the topic ?

erwincoumans, 2020-02-17:
python3 -m pybullet_utils.runServer&
Is that what you were looking for?

mbusy, 2020-02-17:
Hi Erwin, more or less yes! (without a GUI, but I'm guessing that I can use something like pybullet.connect(SHARED_MEMORY_SERVER) to do that).

I was actually wondering if creating several physics servers that way would allow me to run multiple simulations in parallel (using multiple cores)? Thanks again !

erwincoumans, 2020-02-18:
I recommend using pubullet_utils.bullet_client and multiprocessing like this example:

https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_robots/panda/batchsim3_grasp.py

erwincoumans, 2020-02-19:
Sure. One more example using MPI instead with pybullet_client is deep_mimic:
python3 mpi_run.py --arg_file train_humanoid3d_spinkick_args.txt --num_workers 4
mpiexec -n 16 python3 -m pybullet_envs.deep_mimic.DeepMimic_Optimizer

This latter is rather finicky, since Tensorflow and MPI sometimes break their API.

closed 2021-04-26 by erwincoumans