bulletphysics/bullet3 #1643: reset environment causes error (WalkerBase environments)

sungsulim, 2018-04-19:
[image]

When calling test_env.reset(), it throws an error which I'm not sure how to parse. I'm using HalfCheetahBulletEnv-v0.

I have two different instance of the environment, one for training and one for evaluating.
At every N timesteps of training, I do evaluation by running episodes on the test_environment. 
It does not throw an error when doing train_env.reset() at the beginning.

What would be the issue?

benelot, 2018-04-19:
Hi,

Can you build a simple, shareable reproduction case for the problem? It
seems that it does not load the ground plane properly.

sungsulim, 2018-04-20:
`import pybullet as p
import pybullet_envs
import gym

train_env = gym.make('HalfCheetahBulletEnv-v0')
test_env = gym.make('HalfCheetahBulletEnv-v0')

train_env.reset()
test_env.reset()`

This gives the error:
[image]

erwincoumans, 2018-04-20:
There will be conflict, since both environments use a global pybullet module handle, and populate it twice etc.
We solve this in the Minitaur environment by using a local module handle, called BulletClient.
For example [bullet_client.py ](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_envs/bullet/bullet_client.py)and v1 Minitaur using [bullet_client.py](https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_envs/minitaur/envs/bullet_client.py)
Then instead of using p.loadURDF, you use bc.loadURDF: all methods are passed through, and the bullet_client adds the right physicsClientHandle.

ShangtongZhang, 2018-05-11:
I find an easy workaround, if we start each env in a new process, this collision will no longer exists. And I guess this is the reason why OpenAI Roboschool doesn't use the official bullet3, they use a forked branch named `roboschool_self_collision` instead.

erwincoumans, 2018-05-11:
There is no need for Robotschool. The environment needs to be modified to use bulliet_client.py, and then all should work well.

benelot, 2018-05-12:
It is not the reason why roboschool uses a fork. They hack some constants
into the physics engine. That is why roboschool will be outdated pretty
quickly. Check out benelot/pybullet-gym for the most up-to-date gyms. I
made a separate repository to free Erwin from the burden of maintaining the
envs as development goes along.

ShangtongZhang, 2018-05-12:
@benelot Thanks for sharing this, it looks much helpful. I will check it out.

erwincoumans, 2018-05-18:
Fixed here: https://github.com/bulletphysics/bullet3/pull/1690
Now you can create multiple of those environments in the same process (same thread or other thread). The Gym environments from this repository ship with PyBullet, including Ant, Hopper, Humanoid, HumanoidFlagrunHarder, InvertedPendulum etc should be stable and considered the reference version.

benelot, 2018-05-19:
So you again intend to keep all envs up to date and maintained within the
bullet repository? I am again working on making the envs more similar to
MuJoCo. My results will all be in pybullet-gym and depend on pybullet. I
would suggest to keep it that way to make them less coupled and reduce your
maintenance work on them. What do you think Erwin?

erwincoumans, 2018-05-19:
Your  pybullet-gym seems a bit more ambitous with a growing number of environments, Keras support etc. I can't keep up with those developments easily and need to grow my our own environments (Minitaur, KUKA etc) that I'm more familiar with. Still, those Mujoco locomotion and pendulum envs (Ant, Humanoid, Hopper etc) are pretty important and I can manage and became familiar with your initial port of Roboschool that is now in PyBullet and maintain those as part of PyBullet. I think Roboschool is a dead project and making those envs fully compatible (as in weights are exchangable with the Mujoco versions) would be nice, but then the version needs to bump up.

closed 2018-05-19 by erwincoumans