bulletphysics/bullet3 #1644: Controlling position of end effector

jagjeet-singh, 2018-04-20:
I am trying to use setJointMotorControl2 for controlling the kuka_iiwa robotic arm. The control modes provided allow me to control the position, angular velocity, and torques of individual joints. However, I am interested in controlling the coordinates of end effector such that I don't have to deal with individual joint control. Is it possible to do that in the given framework?

JerryLingjieMei, 2018-07-09:
@jagjeet-singh Hi, I am also trying to move the kuka arm to a specific location. Would you please share with me how you use `p.calculateInverseKinematics` to do the work?

erwincoumans, 2018-07-09:
see https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/examples/inverse_kinematics.py and other demos in that folder

closed 2018-07-09 by erwincoumans