bulletphysics/bullet3 #1652: Invalidate broadphase pair cache

SeanCurtis-TRI, 2018-04-25:
Is there a way to invalidate and force recomputaion of the broadphase pair cache?

In my code base, I define a pair-wise broadphase filtering functionality. It is evaluated as individual geometries get added to the broadphase structure (leading to corresponding pairs to be placed in the cache). However, if my filter criteria subsequently evolve, I don't know how to update that cache. I've tried calling `updateAabbs` which I thought *might* work. But it didn't. (That may be a function of the `DBVT_BP_PREVENTFALSEUPDATE`; I'm not sure if it defaults to be defined or not.)

I'd appreciate any guidance.

scpeters, 2018-04-25:
I had this same issue in the past; Erwin recommended calling `cleanProxyFromPairs`, so we do that in gazebo when we move or resize something:

https://bitbucket.org/osrf/gazebo/pull-requests/683/model-resize-in-bullet-mode/diff#chg-gazebo/physics/bullet/BulletLink.cc

I think he made the same recommendation in the following issue: https://github.com/bulletphysics/bullet3/issues/871#issuecomment-262604136

SeanCurtis-TRI, 2018-04-25:
Oops...apparently my ability to search for prior issues is atrocious *and* I have a short-term memory cache that doesn't extend back 18 months. I'd forgotten about that conversation and the recommendations he'd made.

closed 2018-04-25 by SeanCurtis-TRI

scpeters, 2018-04-25:
I've been there

`cleanProxyFromPairs` seems to work for us, though I haven't done much performance profiling