bulletphysics/bullet3 #1654: btConvexHullComputer faces?

JoshKlint, 2018-04-26:
I am trying to use the btConvexHullComputer class to construct a convex hull. I need to retrieve the vertices, edges, and faces. I am confused about how faces are stored. It' looks to be a single integer array with this explanation:

> // Faces of the convex hull. Each entry is an index into the "edges" array pointing to an edge of the face. Faces are planar n-gons
This makes no sense to me. If it is storing an array of edge indices how can there be multiple faces stored? The Edge class is defined inside the btConvexHullComputer class and does not have any information on faces.

JoshKlint, 2018-05-20:
I have it working but when I use this code I get intermittent random crashes, as if some memory is being randomly overwritten somewhere. Can you see anything I am doing wrong?:
`	shared_ptr<ConvexHull> CreateConvexHull(const std::vector<Vec3>& points)
	{
		if (points.size() < 4) return nullptr;

		std::vector<float> f;
		for (int i = 0; i < points.size(); ++i)
		{
			f.push_back(points[i].x);
			f.push_back(points[i].y);
			f.push_back(points[i].z);
		}

		btConvexHullComputer* computer = new btConvexHullComputer();
		if (computer->compute(&f[0], 12, points.size(), 0.0f, 0.0f) < 0.0f)
		{
			delete computer;
			return nullptr;
		}
		
		if (computer->vertices.size() < 4 or computer->faces.size() < 4)
		{
			delete computer;
			return nullptr;
		}

		auto hull = make_shared<ConvexHull>();

		for (int i = 0; i < computer->vertices.size(); ++i)
		{
			hull->AddVertex(computer->vertices[i].m_floats[0], computer->vertices[i].m_floats[1], computer->vertices[i].m_floats[2]);
		}

		int v0, v1;
		for (int j = 0; j < computer->faces.size(); j++)
		{
			auto face = hull->AddFace();
			const btConvexHullComputer::Edge* edge = &computer->edges[computer->faces[j]];
			v0 = edge->getSourceVertex();
			face->AddIndice(v0);
			v1 = edge->getTargetVertex();
			while (v1 != v0)
			{
				face->AddIndice(v1);
				edge = edge->getNextEdgeOfFace();
				v1 = edge->getTargetVertex();
			}
			Assert(face->indices.size() > 2);
		}

		delete computer;

		hull->Build();
		return hull;
	}`

cprogrammer1994, 2018-05-20:
Hi!

I think the collision detection does not require the faces to be computed.
You could use a [quickhull](https://github.com/akuukka/quickhull) implementation to calculate the faces.

erwincoumans, 2018-05-20:
Thanks for digging this up. 

Bullet uses the btConvexHullComputer to create polyhedral features to enable optional separating axis test collision detection (GJK/EPA is enabled by default, which doesn't need those features).

I was re-creating the faces in [btPolyhedralConvexShape::initializePolyhedralFeatures](https://github.com/bulletphysics/bullet3/blob/master/src/BulletCollision/CollisionShapes/btPolyhedralConvexShape.cpp#L92), but perhaps your code is more suitable.

Can you provide a repro case of a crash? Attached is your snippet in a form that compiles and runs fine (zipped).
[main.zip](https://github.com/bulletphysics/bullet3/files/2020247/main.zip)

JoshKlint, 2018-05-20:
I think the problem was caused by a race condition somewhere else in my code, and this just happened to make it more apparent.