bulletphysics/bullet3 #1656: Anchors and Joints between soft bodies and multi bodies

ARoefer, 2018-04-30:
Hello,

I've recently worked on exposing more of the soft body system to pybullet. During that work, I noticed that it's not possible to connect soft bodies to multi bodies using anchors or joints. Is that correct?
My group would like to be able to do that for a research project. I have a time budget for working on this, but I'm not sure how much work it actually is and where to start. 
Could you give me a few pointers?

I'd like to contribute this feature if it's not too time consuming.

JanMatas, 2018-05-10:
Hi,

I have not seen a way to create an anchor between softbody and multibody either. My workaround was to create a tiny rigidbody attached to multibody and then anchor the softbody to the rigidbody. Obviously it would be much better to find a direct way of doing it.

ARoefer, 2018-05-10:
Thank you for confirming that that workaround works, Jan. I thought about trying that, but hoped for a response from Erwin, before going ahead and exposing rigid bodies to pybullet.

ARoefer, 2018-05-15:
In the last couple of days I tried implementing Jan's solution by exposing the creation of rigid bodies to PyBullet and using constraints. (Sadly simply loading the URDF as rigid bodies didn't work.) 
However, I've run into a problem when creating user constraints between rigid and multi bodies.
Video demonstrating the problem: https://youtu.be/cl62iTtHXZg?t=20

If you could point to the code where I can find the implementation for the anchors, I could work on exposing them. The anchor definition in btSoftBody.h only allows for rigid bodies.

erwincoumans, 2018-05-15:
We looked at our previous test, and indeed we also attached a rigid body to the multi body first. So that is the easiest way to get going. There is no anchor implementation between btMultiBody and btSoftBody yet.

>> However, I've run into a problem when creating user constraints between rigid and multi bodies.

A small repro case with URDF would allow someone to help.

ARoefer, 2018-05-15:
Thank you for offering your help! I've put together a small demo "urdf_rigid_body_constraint_bug.py" in this repository: https://github.com/ARoefer/pbsbtest
I work in a ROS environment, but I've removed the dependencies for this demo. However you need to export a "ROS_PACKAGE_PATH" environment variable which points to the folder containing the cloned package. 
You'll also have to use the devel-version of my bullet fork: https://github.com/ARoefer/bullet3.git

erwincoumans, 2018-05-15:
Thanks, but I expected 1 URDF file (with optional assets, OBJ file, MTL, png etc) and  10 lines of regular PyBullet. Why do I need entire repositories and forks?

ARoefer, 2018-05-15:
It seems to be a problem with constraints multi and rigid bodies, which are created in the context of pybullet and this is the environment in which this bug occurs. 
I get that this is quite complicated and fully understand your hesitation. I will just try to dig trough this on my own, or try to put together a smaller example, if I fail.

erwincoumans, 2018-05-15:
There could be an issue with creating a fixed constraint between btMultibody and btRigidBody, attached repro case. It works to attach a multi body cube to KUKA, but not for rigid body cube.
[multibody_rigidbody.zip](https://github.com/bulletphysics/bullet3/files/2006918/multibody_rigidbody.zip)

ARoefer, 2018-05-15:
That seems to be the case. I've exchanged the URDF cube for a rigid body one and this is the result:
https://youtu.be/wLU2y5ztRzc

ARoefer, 2018-05-19:
Hello Erwin,

I should've attached the changed code to my response. I've only changed the line creating the cube. If you'd like me to, I can put together a small pull request for bullet adding the rigid body creator to python. This would allow you to actually run the code yourself and checkout the bug.

[multibody_rigidbody.zip](https://github.com/bulletphysics/bullet3/files/2019815/multibody_rigidbody.zip)

erwincoumans, 2018-05-29:
This PR contains a fix for fixed constraints between btMultiBody and btRigidBody:
https://github.com/bulletphysics/bullet3/pull/1716

ARoefer, 2018-06-04:
Hello Erwin,

I've finally found the time to check out that pull request. The fix works for my scenario. Thank you so much for helping me with this!

Cheers

closed 2018-06-08 by ARoefer