bulletphysics/bullet3 #1672: Port to OpenCL ??

gammastorm, 2018-05-05:
Heya, would be able to port bullet to opencl ???

benelot, 2018-05-05:
Hi!

Would you be able to do it or are others meant to do it? There were tries
to do so in the past, but there were compatibility problems with different
OpenCL drivers as far as I can tell. Another thung that might not always be
simple speed up on GPUs is the resolving of a single island (a heap of
touching bodies basically), since this makes it not parallelizable. You can
resolve multiple islands at the same time though! If you are interested on
working on the OpenCL part, I am sure Erwin Coumans will be glad to give
you some pointers on where to continue.

erwincoumans, 2018-05-05:
The OpenCL port is still therw, you can try it out by running the Bullet Example Browser with the flag --enable_experimental_opencl

Our OpenCL port DOES work for single large islands, using graph coloring. We published some papers/book chapter about this. See the docs folder.

closed 2018-05-08 by erwincoumans