bulletphysics/bullet3 #1673: Crash for btCollisionObject in dynamic world when multi-threading is enabled

przemir, 2018-05-05:
1. My program using single-threaded btDiscreteDynamicsWorld crashed because of btSequentialImpulseConstraintSolver::getOrInitSolverBody method returned -1
It happened because of line:

`if ( btRigidBody* rb = btRigidBody::upcast( &body ) )`

I use CollisionObjects aside from RigidBodies. After I commented line above, single-threaded simulation started to work again.
Similar line is in btSequentialImpulseConstraintSolverMt::getOrInitSolverBodyThreadsafe though.

2. After configuring btDiscreteDynamicsWorldMt with specified btSequentialImpulseConstraintSolverMt program crashes as well.
For btDiscreteDynamicsWorldMt without btSequentialImpulseConstraintSolverMt it works.
I guess (like in the case above) it may have problems when there are pure CollisionObjects inside dynamic world aside from RigidBodies.
To reproduce problem with a small amount of code, I modified sample 'MultiThreadedDemo' a bit, i.e replaced fragment of method MultiThreadedDemo::createStack

from: 
`				btRigidBody* body = localCreateRigidBody( mass, trans, boxShape );
				body->setFriction( 1.0f );
				body->setRollingFriction( gSliderRollingFriction );
`

to: 
`				if(iX % 2 == 0) {
					btRigidBody* body = localCreateRigidBody( mass, trans, boxShape );
					body->setFriction( 1.0f );
					body->setRollingFriction( gSliderRollingFriction );
				}
				else {
					btCollisionObject* collisionObject = new btCollisionObject();
					collisionObject->setWorldTransform(trans);
					collisionObject->setCollisionShape(boxShape);
					m_dynamicsWorld->addCollisionObject(collisionObject);
				}
`

Now it causes crash.
There is btAssert: btBatchedConstraints.cpp:215 (conInfo.numConstraintRows <= 6)

* CMAKE configuration: 32-bit, Debug

lunkhound, 2018-05-07:
That assert in btBatchedConstraints should be removed. I'll make a PR.

lunkhound, 2018-05-10:
With the bad assert removed, I'm not seeing any crash related to btSequentialImpulseConstraintSolver::getOrInitSolverBody using the reproduction steps you provided.
Do you have repro steps for that crash?

przemir, 2018-05-10:
With bad assert removed demo works.

> Note that your collision objects which are not rigid bodies should have either the collision flag CF_STATIC_OBJECT or CF_KINEMATIC_OBJECT set.
I have some objects without those flags indeed. Ok, I will add CF_KINEMATIC_OBJECT flag to them and see if this solve the problems.

Thanks.

przemir, 2018-05-12:
ad1. With CF_KINEMATIC_OBJECT flag added everything works just fine.
ad2. I debugged crash when btDiscreteDynamicsWorldMt is with specified btSequentialImpulseConstraintSolverMt.

I think I found source of problem within btSequentialImpulseConstraintSolverMt::internalCollectContactManifoldCachedInfo:
```
void btSequentialImpulseConstraintSolverMt::internalCollectContactManifoldCachedInfo(btContactManifoldCachedInfo* cachedInfoArray, btPersistentManifold** manifoldPtr, int numManifolds, const btContactSolverInfo& infoGlobal)
{
    BT_PROFILE("internalCollectContactManifoldCachedInfo");
    for (int i = 0; i < numManifolds; ++i)
    {
        btContactManifoldCachedInfo* cachedInfo = &cachedInfoArray[i];
        btPersistentManifold* manifold = manifoldPtr[i];
        btCollisionObject* colObj0 = (btCollisionObject*) manifold->getBody0();
        btCollisionObject* colObj1 = (btCollisionObject*) manifold->getBody1();

        int solverBodyIdA = getOrInitSolverBodyThreadsafe( *colObj0, infoGlobal.m_timeStep );
        int solverBodyIdB = getOrInitSolverBodyThreadsafe( *colObj1, infoGlobal.m_timeStep );

        cachedInfo->solverBodyIds[ 0 ] = solverBodyIdA;
        cachedInfo->solverBodyIds[ 1 ] = solverBodyIdB;
        cachedInfo->numTouchingContacts = 0;

        btSolverBody* solverBodyA = &m_tmpSolverBodyPool[ solverBodyIdA ];
        btSolverBody* solverBodyB = &m_tmpSolverBodyPool[ solverBodyIdB ];

        ///avoid collision response between two static objects
        if ( solverBodyA->m_invMass.fuzzyZero() && solverBodyB->m_invMass.fuzzyZero() )
            break;

        int iContact = 0;
        for ( int j = 0; j < manifold->getNumContacts(); j++ )
        {
            btManifoldPoint& cp = manifold->getContactPoint( j );

            if ( cp.getDistance() <= manifold->getContactProcessingThreshold() )
            {
                cachedInfo->contactPoints[ iContact ] = &cp;
                cachedInfo->contactHasRollingFriction[ iContact ] = ( cp.m_combinedRollingFriction > 0.f );
                iContact++;
            }
        }
        cachedInfo->numTouchingContacts = iContact;
    }
}
```

Shouldn't be 'continue' instead of 'break' in code:
```
        ///avoid collision response between two static objects
        if ( solverBodyA->m_invMass.fuzzyZero() && solverBodyB->m_invMass.fuzzyZero() )
            break;
```

This leaved cachedInfo->numTouchingContacts uninitialized for all remaining manifolds.
After I changed 'break' to 'continue' there was no crash anymore.

lunkhound, 2018-05-12:
Nice! Thank you for tracking that down. 
I agree that 'break' statement looks totally wrong.
I'll fix it.

lunkhound, 2018-05-20:
My fix has been merged now. I removed that if/break statement and replaced it with an assert.

If you have infinite mass bodies which are lacking the correct flags your simulation can suffer performance-wise due to:

- the generating and processing of useless contact manifolds
- unnecessary merging of simulation islands

So the assert I've added is there to warn you about that performance issue.

erwincoumans, 2018-06-10:
Thanks all!

closed 2018-06-10 by erwincoumans