bulletphysics/bullet3 #1681: Failed to create an OpenGL context when trying to launch a pybullet example

trougnouf, 2018-05-11:
Hello,
I tried to launch an example but it fails with the following:
```
$ python enjoy_TF_HalfCheetahBulletEnv_v0_2017may.py 
pybullet build time: May 11 2018 16:36:39
WARN: gym.spaces.Box autodetected dtype as <class 'numpy.float32'>. Please provide explicit dtype.
WARN: gym.spaces.Box autodetected dtype as <class 'numpy.float32'>. Please provide explicit dtype.
WARN: gym.spaces.Box autodetected dtype as <class 'numpy.float32'>. Please provide explicit dtype.
WARN: gym.spaces.Box autodetected dtype as <class 'numpy.float32'>. Please provide explicit dtype.
WalkerBase::__init__ start
WARN: Environment '<class 'pybullet_envs.gym_locomotion_envs.HalfCheetahBulletEnv'>' has deprecated methods '_step' and '_reset' rather than 'step' and 'reset'. Compatibility code invoked. Set _gym_disable_underscore_compat = True to disable this behavior.
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Failed to create GL 3.0 context ... using old-style GLX context
Failed to create an OpenGL context
```

I installed pybullet on Arch Linux using the setup.py script which I packaged as https://aur.archlinux.org/packages/python-pybullet-git/ . I also packaged https://aur.archlinux.org/packages/roboschool-git/ which builds with an old version of bullet3 and I was able to run their examples.

trougnouf, 2018-05-11:
It works on my laptop using the same installation method.
The laptop runs on AMD open-source driver with Wayland while the desktop uses NVIDIA/proprietary driver. They are both up to date and should have approximately the same packages installed.
```
[trougnouf@uryzen:pybullet_envs/examples]$ python testBike.py     (05-11 17:23)
pybullet build time: May 11 2018 17:14:09
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.0 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=X.Org
GL_RENDERER=AMD RAVEN (DRM 3.23.0 / 4.16.7-1-ARCH, LLVM 6.0.0)
GL_VERSION=4.5 (Core Profile) Mesa 18.0.3
GL_SHADING_LANGUAGE_VERSION=4.50
pthread_getconcurrency()=0
Version = 4.5 (Core Profile) Mesa 18.0.3
Vendor = X.Org
Renderer = AMD RAVEN (DRM 3.23.0 / 4.16.7-1-ARCH, LLVM 6.0.0)
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
ven = X.Org
rpy=" 0.0 -0.0 0.0 " xyz=" -0.07000000029802322 0.0 0.6899999976158142
XIO:  fatal IO error 11 (Resource temporarily unavailable) on X server ":0"
      after 694 requests (694 known processed) with 0 events remaining.
```

erwincoumans, 2018-05-11:
The error is reported: Failed to create GL 3.0 context
Are you using a virtual environment, remote desktop connection? You may only be able to run old OpenGL 2 apps. You could modify the code to create a rudimentary OpenGL 2 display, but best is to sort out your graphics drivers.

erwincoumans, 2018-05-11:
You can try to modify testbike.py as follows, and report if you see a graphics window:
p.connect(p.GUI, options="--opengl2")

Note that the OpenGL 2.x implementation is rudimentary and doesn't have all features.

trougnouf, 2018-05-11:
I don't know what happened but it works on the desktop computer after a reboot, using a GL 3.0 context. Thank you for looking into it!

closed 2018-05-11 by trougnouf

DanielTakeshi, 2020-07-22:
@trougnouf  How do you "use a GL 3.0 context" here?

AdityaKapoor74, 2020-08-13:
I am facing the same issue. I am using Ubuntu 18.04 on my VirtualBox. Any updates?

Sicelukwanda, 2020-08-13:
Was facing the same issue on Ubuntu 16.04, in my case the cause was these two lines:

```
export MESA_GL_VERSION_OVERRIDE=3.2
export MESA_GLSL_VERSION_OVERRIDE=150
```

In my bashrc file, everything works fine now.

AdityaKapoor74, 2020-08-14:
I am using virtualbox with Ubuntu 20.04 and whenever I try to run an example, I get the following error:
```

pybullet build time: Aug 14 2020 02:51:12
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Failed to create GL 3.3 context ... using old-style GLX context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=VMware, Inc.
GL_RENDERER=SVGA3D; build: RELEASE;  LLVM;
GL_VERSION=2.1 Mesa 20.0.8
GL_SHADING_LANGUAGE_VERSION=1.20
pthread_getconcurrency()=0
Version = 2.1 Mesa 20.0.8
Vendor = VMware, Inc.
Renderer = SVGA3D; build: RELEASE;  LLVM;
Compile failed:
0:1(10): error: GLSL 1.50 is not supported. Supported versions are: 1.10, 1.20, and 1.00 ES

Segmentation fault (core dumped)
```
Can someone help me with how to rectify this error?

DanielTakeshi, 2020-08-14:
I fixed my error by creating a fake display. There are a number of steps involved but happy to explain in more detail if I can.

AdityaKapoor74, 2020-08-14:
Hey yeah I will be happy to know how you did it?

shanki98, 2020-10-07:
Hi @AdityaKapoor74 , were you able to solve this issue because I have a similar SEGFAULT before it connects to OpenGL

erwincoumans, 2020-10-07:
The OpenGL3 visualizer requires OpenGL 3.x. Most virtual environments don't work well, so it is best to just use DIRECT mode and getCameraImage using the default CPU renderer. Otherwise use a regular non-virtual OS with an up-to-date NVIDIA or AMD display driver.

aamorel, 2020-11-03:
hello @AdityaKapoor74, @shanki98  I have exactly the same problem as you on my VM environment, Ubuntu 20.04.1. 
Did you eventually solve the problem ? When using p.DIRECT, I don't get a graphics window. I want to be able to see what my simulation does. 
Thank you

Robertreido, 2020-12-10:
I am having the same problem on my Nvidia Jetson TX2, running ubuntu 18.04, which has just been rebooted. Have tried what seems to be everything. please help....

Error:
```
pybullet build time: Dec  9 2020 15:39:08
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
X11 functions dynamically loaded using dlopen/dlsym OK!
Failed to retrieve a framebuffer config
```

erwincoumans, 2020-12-10:
It should work on TX2 (have it running here), probably an issue with the display drivers. Also, I assume you run it with a display monitor attached, right (it won't work otherwise). What does nvidia-smi tell?

Robertreido, 2020-12-10:
@erwincoumans this is all i get when running 'watch nvidia-smi'. And yes i am running with a display monitor attached.
```
sh: 1: nvidia-smi: not found

```

erwincoumans, 2020-12-10:
OpenGL 3 won't work well without a proper installed NVIDIA driver.

Robertreido, 2020-12-10:
Could you please point me in the right direction on how to do that? i am feeling quite out of my depth sorry

Robertreido, 2020-12-11:
@erwincoumans  would really appreciate it please, really stuck at the moment

CHbloom, 2023-02-11:
> I am using virtualbox with Ubuntu 20.04 and whenever I try to run an example, I get the following error:
>
> ```
>
> pybullet build time: Aug 14 2020 02:51:12
> startThreads creating 1 threads.
> starting [...]

have you solve it?I met the same problem,could you help me.

CHbloom, 2023-02-11:
> @erwincoumans would really appreciate it please, really stuck at the moment

have you solve it?I met the same problem,could you help me.