bulletphysics/bullet3 #1691: Failed to init some gym environments

ShangtongZhang, 2018-05-20:
```
    import pybullet_envs
    game = 'ReacherBulletEnv-v0'
    game = 'ThrowerBulletEnv-v0'
    game = 'HalfCheetahBulletEnv-v0'
    env = gym.make(game)
    env.reset()
```

```
Traceback (most recent call last):
  File "/opt/project/examples.py", line 449, in <module>
    env.reset()
  File "/usr/local/lib/python3.5/dist-packages/gym/wrappers/time_limit.py", line 44, in reset
    return self.env.reset()
  File "/usr/local/lib/python3.5/dist-packages/pybullet_envs/gym_locomotion_envs.py", line 28, in _reset
    r = MJCFBaseBulletEnv._reset(self)
  File "/usr/local/lib/python3.5/dist-packages/pybullet_envs/env_bases.py", line 67, in _reset
    s = self.robot.reset(self._p)
  File "/usr/local/lib/python3.5/dist-packages/pybullet_envs/robot_bases.py", line 124, in reset
    self.robot_specific_reset(self._p)
TypeError: robot_specific_reset() takes 1 positional argument but 2 were given
```

```
root@0afadd8d2e39:/workspace# pip freeze | grep 'bullet'
pybullet==1.9.8
```

Some environments work well (e.g., Ant) but some failed. Maybe it has sth to do with https://github.com/bulletphysics/bullet3/pull/1690

erwincoumans, 2018-05-20:
It is fixed in the pull request: https://github.com/bulletphysics/bullet3/pull/1692
will upload pybullet 1.9.9 soon
Thanks for the report!

ShangtongZhang, 2018-05-21:
Thanks for your quick response, however some games are still wrong in the new version.
```
    import pybullet_envs
    games = [spec.id for spec in gym.envs.registry.all() if spec.id.find('Bullet') >= 0]
    bugs = []
    for game in games:
        try:
            env = gym.make(game)
            env.reset()
            env.step(np.random.random(env.action_space.shape))
        except Exception as e:
            print(e)
            bugs.append(game)
    print(bugs)
```
Most are
```
episode_restart() takes 1 positional argument but 2 were given
```

erwincoumans, 2018-05-22:
More fixes in this PR: https://github.com/bulletphysics/bullet3/pull/1698
Thanks for the report!

closed 2021-04-26 by erwincoumans