bulletphysics/bullet3 #1697: No module named 'pybullet_envs.bullet.simpleHumanoidGymEnv

joabim, 2018-05-22:
Hi,

Running
```
 from pybullet_envs.bullet import minitaur_gym_env
```
yields the following import error
```
ImportError                               Traceback (most recent call last)
<ipython-input-1-a096efc36562> in <module>()
----> 1 from pybullet_envs.bullet import minitaur_gym_env

~/dev/bullet3/examples/pybullet/gym/pybullet_envs/bullet/__init__.py in <module>()
      4 from pybullet_envs.bullet.racecarGymEnv import RacecarGymEnv
      5 from pybullet_envs.bullet.racecarZEDGymEnv import RacecarZEDGymEnv
----> 6 from pybullet_envs.bullet.simpleHumanoidGymEnv import SimpleHumanoidGymEnv
      7 from pybullet_envs.bullet.kukaGymEnv import KukaGymEnv
      8 from pybullet_envs.bullet.kukaCamGymEnv import KukaCamGymEnv

ImportError: No module named 'pybullet_envs.bullet.simpleHumanoidGymEnv'

```
When inspecting the source, it seems like the `simpleHumanoidGymEnv` module is missing under `pybullet_envs/bullet/`. Is this some legacy environment that is no longer used?

erwincoumans, 2018-05-22:
Yes, it is a removed environment, just delete the line in the __file__, I'll fix it in our repo asap.
Thanks for the report!

nickrobinson251, 2018-05-29:
The removed environment also means all `python -m ...` examples listed under [RL Gym Envs documentation](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#heading=h.wz5to0x8kqmr) in the quickstart guide now fail with `ModuleNotFoundError: No module named 'pybullet_envs.bullet.simpleHumanoidGymEnv'`