bulletphysics/bullet3 #1703: Error on Box to Box GJK algorithm

AndreaCatania, 2018-05-23:
This is a bug that I found on Godot: https://github.com/godotengine/godot/issues/18794
This is the reproducible demo [BasicExample.zip](https://github.com/bulletphysics/bullet3/files/2030925/BasicExample.zip)

This problem is similar to the issue #1623 where the returned penetration is positive.
I've noticed that by setting static body extension from `128, 1, 128` to `1,1,1` the problem disappear.
Also the problem doesn't show up if I remove the line with `kinShape->setMargin(0);` 

This seems to happen only with Box to Box algorithm, Also is mandatory for me set a margin of 0.

Thanks you in advance

AndreaCatania, 2018-05-26:
Erwin thanks you! it fix the problem!! If you merge it I will import the new version of bullet in godot

erwincoumans, 2018-06-11:
I need to sort out a few more things related to this collision detection and get back to you when it is done.

AndreaCatania, 2018-06-11:
Ok let me know thanks you

AndreaCatania, 2018-07-07:
Hi Erwin, any advance on this?

AndreaCatania, 2018-07-31:
Hi Erwin, probably you are really busy, I want just remember you this issue. Let me know thanks you!!

erwincoumans, 2018-07-31:
Yes, it is on my todo, once returned feom vacation.

AndreaCatania, 2018-07-31:
Ok thanks you, have nice vacation

erwincoumans, 2018-08-15:
I've been looking into the problem, it is a mix of issues. Need a bit more time, working on it. In the meanwhile, avoid 0 margin, use smaller shape size differences, use double precision, strict float (instead of fast).

AndreaCatania, 2018-08-15:
In this moment I can't apply any of these, except for 0 margin. But I think that I can wait the fix, so Thanks you for looking on it

erwincoumans, 2018-08-16:
Can you please check the attached file and see if that fixes the issue(s)?

+ Please never set the collision margin to zero (preferably leave it default). 
+ Don't make it easy for the user to set the margin to zero, since it will lead to incorrect collisions and slow performance.
+ Don't set this setEqualVertexThreshold to zero, please remove the line:
~~~~~~~~~~
gjk_simplex_solver->setEqualVertexThreshold(0.f);
~~~~~~~~~~
[btGjkPairDetector.zip](https://github.com/bulletphysics/bullet3/files/2292248/btGjkPairDetector.zip)

Thanks for the report and testing!

erwincoumans, 2018-08-31:
Some improvements are checked in now. Please provide some feedback.
https://github.com/bulletphysics/bullet3/pull/1850/commits/ee9fca8c29149a28769d1128a22155ed30ad7826

closed 2018-08-31 by erwincoumans

AndreaCatania, 2018-08-31:
Sorry if I didn't answer yet, but was a busy period.

AndreaCatania, 2018-08-31:
It works Thanks you!