bulletphysics/bullet3 #1705: when installing from source, pybullet found, but not pybullet_envs

keithmgould, 2018-05-23:
I followed directions to install from source via the [pybullet quickstart](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#) guide.

In particular, I added:
```
export PYTHONPATH = /your_path_to_bullet/build_cmake/examples/pybullet
```
so now python can find the pybullet module.  However it can not find pybullet_envs. Do I need to add another location to the PYTHONPATH? If so what?  Also, would be nice if this second path was mentioned in the quickstart.

Thank you!

PS: Had to install from source because `sudo pip3 install pybullet` was giving that silly "thread-local storage is not supported for the current target" and I could not figure out how to get around it. :/

erwincoumans, 2018-05-23:
You can find the pybullet_envs in the bullet/examples/pybullet/gym folder, and the code should work in-place (just open a terminal in that place). You can also copy those folders manually into your python distribution (in the lib/site-packages).

You probably want to solve the 'thread-local storage is not supported for the current target' problem, google for it, install a newer llvm, or newer xcode etc.

closed 2018-05-23 by erwincoumans

keithmgould, 2018-05-23:
Thank you, running python from the gym folder let me run the examples.

As far as the thread-local problem, I did google, and did install a newer compiler, neither of which helped, as I was unsure how to tell PIP to use the compiler. As mentioned, I tried `CC=/newer/compiler/path CXX=/newer/compiler/path PIP3 etc...' and that did not help.

Also, I have a pretty new LLVM, right?
```
$ llvm-gcc --version
Apple LLVM version 9.1.0 (clang-902.0.39.1)
Target: x86_64-apple-darwin17.5.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
``` 

Is there a solution somewhere that you know works that is explicit?

Thank you again!

erwincoumans, 2018-05-25:
Could you try attached setup.py, it tries to use thread_local in C++11.

[setup.zip](https://github.com/bulletphysics/bullet3/files/2038929/setup.zip)

erwincoumans, 2018-05-25:
Could you try attached setup.py, it tries to use thread_local in C++11.
[setup.zip](https://github.com/bulletphysics/bullet3/files/2039043/setup.zip)

amisha2019, 2020-06-27:
@erwincoumans hello i've installed pybullet successfully and it is even show when i run pip show pybullet. but when i run my program it shows error: module not found.
please help me with this.

erwincoumans, 2020-06-27:
@amisha2019  you need to create symbolic links in the right Python directory to pybullet_envs, pybullet_data and pybullet_utils. If that doesn't work, just use python setup.py develop in the root of Bullet (or pip install pybullet)

amisha2019, 2020-06-27:
Thanks, the module is installed now.