bulletphysics/bullet3 #1709: B3_USE_DOUBLE_PRECISION

eastskykang, 2018-05-26:
Hello, 

I am using Bullet Physics Robotics API.

I want to use double precision for API so I am looking for cmake option that turn on "B3_USE_DOUBLE_PRECISION" but could not find from CmakeList files.

Is it just omitted?

erwincoumans, 2018-05-27:
It is better to switch to btVector3/btQuaternion in that API to support consistent double precision, here is the pull request: https://github.com/bulletphysics/bullet3/pull/1710
Thanks for the report. If there are missing API's that need to be exposed for leggedrobotics/SimBenchmark, please report them in a separate issue.

erwincoumans, 2018-05-27:
API now uses btVector3 etc, so cmake with -DUSE_DOUBLE_PRECISION=ON should do the job:
https://github.com/bulletphysics/bullet3/pull/1710

closed 2018-05-27 by erwincoumans