bulletphysics/bullet3 #1714: Null Space Inverse Kinematics has problems with fixed joints

tomsilver, 2018-05-28:
I believe the issue found and fixed in Inverse Dynamics with fixed joints (https://github.com/bulletphysics/bullet3/issues/1379) may also be present in Null Space Inverse Kinematics.

In particular, `lowerLimits`, `upperLimits`, `jointRanges`, `restPoses` are expected to have lengths equal to the number of joints by `calculateInverseKinematics` : see e.g.  https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/pybullet.c#L8120

But once passed to bullet, it looks like the number of dofs is expected instead: e.g. https://github.com/bulletphysics/bullet3/blob/abeae7e1e7e23d5ba8e840e98b46470d4162a73b/examples/SharedMemory/PhysicsClientC_API.cpp#L4001

Unfortunately padding the arrays until they are length numJoints (as suggested in the previous issue) doesn't seem to fix this one.

I am still finding my way around the code base so apologies if I've read this wrong. Thanks again for your time and help!

tomsilver, 2018-05-30:
@erwincoumans thanks for your reply!

The URDFs and meshes are too large to paste here, is a dropbox link okay?
https://www.dropbox.com/s/lb3ukdwnkeja6m4/pybullet_ik_test.zip?dl=0

In the example, using Inverse Kinematics without null space arguments behaves as expected:

[image]

But with null space arguments based in, things go haywire:

[ikNumJoints.mov.zip](https://github.com/bulletphysics/bullet3/files/2055728/ikNumJoints.mov.zip)

In the provided PyBullet script, the two examples can be reproduced by switching between `useNullSpace=True` and `useNullSpace=False` in the call to `accurateIK`.

erwincoumans, 2018-05-31:
Thanks, looks like a nice model, I'm looking into it.

erwincoumans, 2018-05-31:
close it for now, please re-open if there is still an issue.

tomsilver, 2018-06-01:
@erwincoumans thanks so much for taking the time to dig into this, I really appreciate it.

I think the issue should be tentatively reopened (I do not have privileges to reopen myself). I am suspicious that the "Null Space mode" for IK is only activated when the arrays are equal to the number of joints (fixed joints included).

So I think what happens when you skip the fixed joints is that the arrays are effectively ignored.

I have verified that if you pass in random or arbitrary arrays of length `numDofs` to the accurate IK calculator, the result is identical if you pass in arrays with the correct joint information. The IK behavior only changes when the arrays if of length `numJoints`. My suspicion is also backed up by the C++ code I cited in the original issue.

Let me know if there's any other tests I can write to verify whether Null Space mode is being used or not.

Thanks again!

erwincoumans, 2018-06-01:
It is fixed now. Picking the sliding joint adds a challenge, better to pick joint 48 for Baxter.
Note that you need to be careful with setting the joint ranges and limits, since there are conflicting constraints it can easily go unstable. For now, I made the joint limits/ranges very wide.
See  updated demo https://github.com/erwincoumans/pybullet_robots/blob/master/baxter_ik_demo.py
and sync to latest source at github (or wait for pip pybullet 2.0.2)

erwincoumans, 2018-06-01:
Video is being uploaded here:
https://www.youtube.com/edit?o=U&video_id=AQ2ACC05uzU

tomsilver, 2018-06-01:
@erwincoumans code and video look great, thanks so much!

Side note -- it would be great if the quickstart guide could be updated to disambiguate between numDofs and numJoints whenever applicable.

Thanks again! Closing.

closed 2018-06-01 by tomsilver

tomsilver, 2018-06-01:
> The quick start guide
> <https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#>
> did mention the range [0..nDof] in the calculateInverseKinematics. I added
> a few more notes to explain it.

Appreciate it!

wltjr, 2024-06-17:
Hello, I know this is years old, but I have a use for PyBullet and Baxter, and using [baxter_ik_demo.py](https://github.com/erwincoumans/pybullet_robots/blob/master/baxter_ik_demo.py) as a reference, starting point. It seems if I expand to 2 arms from 1, the nullspace issue comes up again, both arms vibrate a lot, but if I set nullspace to < 0.5 the threshold, they stop freaking out and movement is normal. For my purposes, I am not sure I care about nullspace, just wanted to mention the issue.

I duplicated the [3 target parameters](https://github.com/erwincoumans/pybullet_robots/blob/master/baxter_ik_demo.py#L158), and added in a 2nd call to accurateIK and setMotors using the [ID of the other arm endpoint](https://github.com/erwincoumans/pybullet_robots/blob/master/baxter_ik_demo.py#L37), returning 2 ids from setUpWorld.