bulletphysics/bullet3 #1720: vs2017 - building libs with cmake "double precision" - unresolved public btConvexHullShape::btConvexHullShape

PcChip, 2018-05-31:
When I build with double precision enabled, I get an unresolved external btConvexHullShape::btConvexHullShape  
  
rebuilding the libs without DP, it works as expected

erwincoumans, 2018-05-31:
Odd, we build with cmake and double precision all the time, and no unresolved symbols.

Did you try build_visual_studio_vr_pybullet_double_cmake.bat or
build_cmake_pybullet_double.sh

PcChip, 2018-05-31:
I don't think I used the batch files, I opened the program called "CMake 3.10.0-rc5 ,
source: c:/bullet3-master
where to build binaries: c:/bullet3-master/vs_2017
configure , selected some options, generate -> open project -> set to release -> build entire solution

then closed that solution, opened my current project, added the .lib files along with the lib include directory of: bullet3-master\vs_2017\lib  
  
I'm still new at this, and could be doing it all wrong

AndresTraks, 2018-05-31:
Your own project needs to have BT_USE_DOUBLE_PRECISION defined when including the Bullet headers.
Otherwise the project will expect single precision methods in the libs.

erwincoumans, 2018-05-31:
Thanks Andres, that is likely the issue. Let's close it (reopen if it doesn't fix it)

closed 2018-05-31 by erwincoumans