bulletphysics/bullet3 #1721: Striker and Thrower xml not loading properly

benelot, 2018-05-31:
There are some problems with loading the thrower.xml and the striker.xml with pybullet:
(https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_data/mjcf/striker.xml, https://github.com/bulletphysics/bullet3/blob/master/examples/pybullet/gym/pybullet_data/mjcf/thrower.xml)
-The thrower has some kind of fingers defined in the xml, however, they do not show up in the simulation.
-The striker xml contains a goal element, but it does not show up as 'goal'. Interestingly, the 'coaster' element shows up. Furthermore, the goal element is defined as 'free' to make it freely movable on the ground, which makes it float around strangely on the ground by itself and it falls half-way through the ground plane.

Any more details I should provide?

erwincoumans, 2018-05-31:
If you want to help out, please consider submitting a patch to fix this. The MJCF importer is here:
https://github.com/bulletphysics/bullet3/blob/master/examples/Importers/ImportMJCFDemo/BulletMJCFImporter.cpp

Having a new port of the Ant and Half Cheetah (to start with) that behaves more similar to the original Gym MuJoCo environments (in addition of the existing Roboschool ports) would be higher on my wish list though.

benelot, 2018-05-31:
For now, it is not on my list either. I just wanted to point out that there
seems to be some discrepancy between the mujoco mjcf importer and the
pybullet mjcf importer. So maybe at some point it can be fixed. But next up
is definitely the Gym MuJoCo envs. I trained some agents on the original
envs and now I am about to build the envs in pybullet. If I may call you in
at some point to help me decrypt some of the mujoco internal values they
are using as observations (basically stuff below --main inputs and outputs
of the computation: http://www.mujoco.org/book/source/mjdata.h)
, then that would be great. Some of them are needed to make similar
observations from pybullet to the agent.

closed 2021-04-26 by erwincoumans