bulletphysics/bullet3 #1726: Multiple Visuals in Link only use single material

keithmgould, 2018-06-01:
Not sure if this is expected behavior. Ideally each visual is rendered with the material tagged within.

Right now I'm getting around this with a hack: making the second visual an entire different link (with fixed joint). In other apps the visuals render differently, so I think this is a pyBullet issue.  

In the below example, the entire bot is rendered red. However the body is blue.

```xml
<?xml version="1.0"?>
<robot name="bot">
  <material name="blue">
    <color rgba="0 0 0.8 1"/>
  </material>

   <material name="red">
    <color rgba="0.8 0 0 1"/>
  </material>

  <link name="body">
    <visual>
      <geometry>
        <box size="0.1016 0.0254 0.6604"/>
      </geometry>
      <origin xyz="0 0 0.3302" />
      <material name="blue"/>
    </visual>

    <visual>
      <geometry>
        <cylinder length="0.17" radius=".01"/>
      </geometry>
      <origin rpy="0 1.57 0" xyz="0 0 0.01905"/>
      <material name="red"/>
    </visual>
  </link>
</robot>
```

Just in case here is the tiny script to run:

```python
import pybullet as p
import time
import pybullet_data

physicsClient = p.connect(p.GUI)
p.setGravity(0,0,0)
planeId = p.loadURDF("simple.urdf")

for i in range (10000):
    p.stepSimulation()
    time.sleep(1./240.)

p.disconnect()
```
Here is the all red bot:

[image]

erwincoumans, 2018-06-01:
Yes, it is a limitation of the PyBullet and its OpenGL debug visualizer, so you need to split visual shapes with multiple materials into separate links. We have a tool to do this for OBJ files (obj2sdf).

closed 2018-06-01 by erwincoumans

davidgolub, 2018-10-11:
Thanks for your help. How do you use obj2sdf?

erwincoumans, 2018-10-11:
./obj2sdf <input.obj>

Use premake4 to compile obj2sdf executable (works on Windows, Linux and Mac OSX). for example on Linux:
~~~~~~~~~~
git clone https://github.com/bulletphysics/bullet3
cd bullet3/build3
./premake4_linux64 gmake
cd gmake
make -j8
#now the obj2sdf executable is in bullet3/bin folder
~~~~~~~~~~