bulletphysics/bullet3 #1737: btCompoundShape change child  after it attached to rigidbody

gl3336563, 2018-06-07:
hello! I  found some problem when I change the child of btCompoundShape(add or romove),  if it attatched 
to a rigidbody have problem, if  not attatched to a rigidbody is Ok. but  i want to dynamic change child  weather it have attached .  I know  change  a shape must remove rigidbody from the  world and then add again and  should compute setMassProps   and calculateLocalInertia.  I have do this. but still not work。Maybe some special thing need to do with btCompoundShape. Maybe my bullet verison is a litte old 2.82. Or I must destroy old btCompoundShape and new another btCompoundShape to set rigidbody again? I know this can work.

BluePrintRandom, 2018-06-07:
I use unparent / transform / reparent each frame.

BluePrintRandom, 2018-06-07:
> Please zip a small modified Bullet example to reproduce your problem.

I think what he is trying to do is have the center of mass change when a
compound parent occurred, (I could be wrong*)

I have done this manually in bge using a empty physics shape and parenting
all children, moving the center of mass obj, parent all child shapes to
empty(compound root) and setting it's mass to reflect its children.

gl3336563, 2018-06-08:
Oh! Thanks to reply me @BluePrintRandom  and  @erwincoumans. I am sorry my english is bad. In fact I am use the javascript version of bullet。 it work great. I will try the C++ sample later. 
I found problem now. if I setLocalScale of btCompoundShape is not (1,1,1).for example is (3,3,3) and then addChild(childTransform,childShape)。the physic collider is small(have problem). if focre set setLocalScale of btCompoundShape(1,1,1) and then addChild(childTransform,childShape),and revert LocalScale(3,3,3) is OK(No problem).
In fact. I think
1. btCompoundShape.setLocalScale(3,3,3)
2.btCompoundShape.addChild(childTransform,childShape)
And
1. btCompoundShape.setLocalScale(1,1,1)
2.btCompoundShape.addChild(childTransform,childShape)
3.btCompoundShape.setLocalScale(3,3,3)
should same,because btCompoundShape.setLocalScale function is set the parent's LocalScale and childTransform is the child's local transform  they are indepent.  two ways world Boud should same.
May I am not deep understand it.

closed 2021-04-26 by erwincoumans