bulletphysics/bullet3 #1738: cannot import pybullet in python after build windows 10

berwul, 2018-06-07:
Hallo! 
I followed the guide on the start page. I did the following steps. 

1. Ran following file
build_visual_studio_vr_pybullet_double.bat

2. Opened 0_Bullet3Solution.sln in visual studio 2017. 

3. Retargeted Project (default settings in visual studio ->  Windows SDK version: 10.0.16299.0 plattform Toolsets: v141.) 

4. Build the solution. 

The build went without any errors. However when I try to import pybullet in python it can't find pybullet? The installation guide doesn't give any guide on what to do? Did I miss some obvious step? :) 

I also tried to install the via pip (pip install pybullet). This failed with an error.

erwincoumans, 2018-06-07:
~~~~~~~
I also tried to install the via pip (pip install pybullet). This failed with an error.
~~~~~~~
This is the main way of installing. What error?

Manually compiling the pybullet plugin is complicated and only for developers who know what they are doing. pip is the way to go.

Python needs to know where to find the plugin, and the name needs to be pybullet.pyd file (instead of the name ending with dll).
Usually I make a symbolic link manually in the c:\python-x.y.z\dll folder, from pybullet.pyd to the dll in the Bullet/bin folder.
Another way is to rename the dll to pybullet.pyd and then add an environment variable PYTHONPATH to point to the folder.

berwul, 2018-06-07:
Hallo Erwin! 

It outputs "Cannot open include file: 'stdint.h': No such file or directory.". 

I've attached a printscreen of the error below. 
[image]

Any idea whats wrong by installing by pip? 

I tried the approach by adding a symbolic link. I did the following.

1. In bullet3-master\bin rename pybullet_vs2010.dll to pybullet.pyd.
2. Add symbolic link to previous file in C:\Python27\DLLs.

When I then tried to import it in python it seemed to work! Does the steps seem right to you? 

Thanks for your help :)

erwincoumans, 2018-06-07:
Do you use the full version of Visual Studio, or some crippled python specific version?
I've seen reports of people who use the python-specific cripped version that has no stdint.h (what were they thinking when removing it?!?). 

See https://github.com/bulletphysics/bullet3/issues/1708

berwul, 2018-06-08:
I have Visual Studio 2017 installed on my computer. Is my python tied to it? I don't know which version my python is tied to. 

Thanks for pointing out the thread issue 1708. However I don't really understand your answer. Do you mean I should just compile it by using Visual studio as I've done previously. 

Thanks for your help!

erwincoumans, 2018-06-08:
If you follow the fix in 1708, you can do "pip install pybullet", and all good to go. You have to sort out the lack of stdint.h issue first.

berwul, 2018-06-08:
Thanks it works now! Thanks for your help :)

closed 2021-04-26 by erwincoumans