bulletphysics/bullet3 #1739: Contact point persists when objects are far apart.

kermado, 2018-06-07:
I have a simple scene containing two boxes. I am looking at the contact points between them and I frequently see one lingering contact point despite the objects not being in contact.

[image]

I have been able to track this down to how contact points are reused. Initially, as the first box just begins to intersect the second, all contact points have normals pointing to the left since this is the direction of least penetration (all as expected).

Then, at a later point, the direction of least penetration changes so that all the normals should be pointing upwards. At this point, the insertion index for two of the contact points determined inside of `btManifoldResult::addContactPoint()` is -1 for two contacts because the minimum distance is greater than the contact breaking threshold. All fine so far.

For those two contact points we call `btPersistentManifold::addManifoldPoint()`, which obtains the insertion index from `btPersistentManifold::sortCachedPoints()`. This function retains the contact point with the greatest penetration depth. And herein lies the problem...

We retain one of the two contact points with the normals pointing to the left. This will from now on always have the greatest penetration distance, which will continue to grow in magnitude. The `btPersistentManifold::refreshContactPoints()` function will then never remove the contact point. The contact point only goes away when the broadphase determines that the two boxes are definitely not intersecting.

erwincoumans, 2018-06-07:
The persistent contact manifold heuristic is meant for dynamics, where deep penetrations are resolved by pushing back the object. You seem to not use dynamics, but slide the box 'through' the other box? In that case, you may want to just manually clear the contact cache every frame.

kermado, 2018-06-07:
Thanks for the explanation - makes sense.
I'll clear the manifold in the collision algorithm `processCollision()` functions if one of the objects has `btCollisionObject::CF_NO_CONTACT_RESPONSE` set.

closed 2018-06-07 by kermado

onefifth, 2019-01-21:
It's bumping an old thread, but we ran into an essentially identical issue and I wanted to see if I could get some clarification on the ideal/correct solution.

We have a btCharacterController colliding with a btCollisionShape that we're using as a trigger volume. We aren't doing resolution between the two groups, only listening to the start and end collision callbacks. We've found that the persistent manifolds linger well after the character controller has completely exited the trigger volume as long as the direction of travel is the inverse of the calculated collision normal, thus failing to dispatch a "collision end" callback. The collision end is _eventually_ fired when the character controller changes direction of motion or the broadphase check determines there's no collision. (character controller exits the AABB of the trigger volume)

I understand why this is happening based on the rules for breaking a persistent manifold, but am wondering what the recommended/efficient way of getting this kind of overlap information would be.