bulletphysics/bullet3 #1740: PyBullet URDF use self collision exclude parent not working

ARoefer, 2018-06-08:
Hello,

I noticed that the URDF_USE_SELF_COLLISION_EXCLUDE_PARENT flag doesn't seem to be working anymore in the current version of PyBullet. URDF_USE_SELF_COLLISION seems to work though. 

Here is a photo showing two KUKAs initialized with the same joint configurations using the different flags. The configuration puts them in a self-intersecting configuration, which gets resolved for the one using the regular self collision flag (right) but not the other one.
[image]

Code to recreate the problem:
[self_collision_exclude_parent_bug.zip](https://github.com/bulletphysics/bullet3/files/2084270/self_collision_exclude_parent_bug.zip)

erwincoumans, 2018-08-31:
Need to look into it. Unfortunately the project lacks resources, so it can take a while.

erwincoumans, 2019-01-31:
Until we get to this, you can achieve the same (and more fine-grained) using collision filters.
See 'setCollisionFilterPair' in the [PyBullet Quickstart Guide](https://docs.google.com/document/d/10sXEhzFRSnvFcl3XxNGhnD4N2SedqwdAvK3dsihxVUA/edit#).

closed 2021-04-26 by erwincoumans